terjeio/ioSender #363: Settings resore - errors

boboxx, 2024-02-29:
When doing a restore of grblhal config using ioSender (latest) there are lots of errors.

Settings:
$170=0.000 returned error:53
$172=0.000 returned error:53
$370=2 returned error:52
$395=1 returned ALARM:14

terjeio, 2024-03-01:
This has (mostly) nothing to do with ioSender.
Different grblHAL configurations may have different settings available, e.g. $170 is related to backlash If backlash is not enabled in grblHAL an attempt to set the $170 will result in an error.

$395 sets the default spindle, if the number of spindles or types of spindles has been changed in the configuretion this can lead to stiuations llike his where an alarm is raised. Alarm 14 is due to communication failure with a VFD spindle.

An issue that could be resolved in ioSender is that some settings are interdependent and might be numbered in the wrong order, this can be solved by reorder the sending of settings to the controller. Or by restoring them twice. And for some of the errors the sender may query the controller about its configuration and not try to set them if not available. I guess this could be done by comparing the settings reported from the controller with the restore file content...

closed 2024-03-13 by boboxx

terjeio, 2024-03-14:
New [edge versions](http://io-engineering.com/downloads) are now up that will count mismatches between the controller and the backup file for settings not supported at all and report the count when the restore is completed. Note that some settings may have different number of flags (bits) available, these may still be causing an error message to be shown,