terjeio/ioSender #365: Basestream is only available if port is open

zurdoid123, 2024-03-08:
Hi again, new issue, After fixing my EMI a while back, seemingly coming from the router. I moved the cable to the opposite side from the controller and it have been working fine. I have since then ran the machine without problems. But yesterday it started making trouble again.

Iosender sops sending commands to the controller (SKR pr1.2) and after a hile it just stops moving. When I then click Stop an error comes up saying that BaseStream is only available if port is open. It seems to happen random times, no other machines running in the workshop so it cant be from their noise. 

I tried yesterday finally hooking the controller up to its own PSU, instead of being powered from the USB, but no change. Still the same problem today.. 

I have turned off the energy saving function for the usb ports in device manager. 

Any tips?

zurdoid123, 2024-03-08:
I now read that someone had a similar problem and it got resolved from him disconnecting wired his mouse so I tried it and have now had the machine running for a while. It still stops from sending but it doesn't give any error or close the sender when I click on stop. I can then start again from a suitable row and continue the job. 

So in a way it got better but still wierd.

terjeio, 2024-03-08:
The error is due to the connection beeing severed, either at the controller side or at the PC side.
When the controller is powered from its own PSU can you unplug/reinsert the USB cable and reconnect? If so the controller has not crashed and the issue might be due to the USB stack(s) somehow beeing corrupted.

> It still stops from sending but it doesn't give any error

Odd, I have never seen that before. IMO again something (EMI?) messing with the USB connection. Have you tried with a USB cable with a [ferrite](https://en.wikipedia.org/wiki/Ferrite_bead) on it?

zurdoid123, 2024-03-08:
Yeah its wierd, especially now when it actually doesn't disconnect but just stops sending commands. 

I'm currently running a finishing pass after finishing the roughing pass, it stopped first after a minute and have now been going strong for about 30min..

I added snap on ferrite on the usb as I couldn't seem to find cables with it built in in my city. That allowed me to run the table saw or mitre with out problems, before it stopped as soon as I started either one of them.

 But now, since yesterday, its been stopping much more frequently than even before I moved the router cable to the other side of the table.

phil-barrett, 2024-03-08:
If your problem is EMI, there are many ways for it to enter your machine. Moving cables is marginally effective, at best.   Here is a [general overview of methods to apply](https://www.grbl.org/single-post/emi-science-voodoo-or-doodoo). 

I highly recommend the following steps at minimum:

1. earth ground your machine
2. switch to ethernet for communications
3. isolate all control inputs (limit switches and so on)
4. use shielded cable and earth ground the shields.

Make sure grounding is a star configuration - don't daisy chain grounds.

zurdoid123, 2024-03-09:
Well for me moving the cable made quite a big difference, didn't have any problems for 2 weeks. But yeah for sure it's not the proper solution or enought. 

I'm in an old sovjet industrial building and grounding is sparse to say the least. I can't really drive a stick in the ground as the whole area around is covered in concrete... But plumbing might work, pipes is most likely old cast iron. I will for sure try grounding everything to the pipes, didn't think about that.

Swaping to a ethernet cable is something I have been thinking about, wasn't sure but now I will for sure give it a try.

What do you mean with insulating all the control inputs? I'm using shielded cables on all the switches and they have never triggered unnecessarily. I have no other inputs, everything is on the computer for now.

I'll read through the guide you linked aswell, thanks!

phil-barrett, 2024-03-09:
> What do you mean with insulating all the control inputs? I'm using shielded cables on all the switches and they have never triggered unnecessarily. I have no other inputs, everything is on the computer for now.

Isolating, not insulating. Use things like opto-isolators to prevent EMI following the wires into the control circuitry.

zurdoid123, 2024-03-13:
I will try changing it to a ethernet cable tomorrow. 

I have now grounded the whole machine, turns out the pipes running through my shop indeed is to ground. Tested with a bulb. It still didn't change alot though. The power was very spotty today though, have actually had two outages in the last few weeks, first ones in since I moved in 3 years ago. 

These opto insulators would then be on all the cables? I will investigate that aswell, thanks!

Today it actually disconnected without anything running, no machine at all. This made me think that mains could be the culprit? I know that the voltage tends to be a bit up and down in the building, could this be a thing? Any ideas how to work around such a problem? I was thinking UPS but then read that they don't condition but only switch on when power dies so I guess that wouldn't really help.

phil-barrett, 2024-03-13:
Optos won't work on high speed stuff. Use for limit switches/sensors and input controls like cyclestart/feedhold.

Bad power is going to be challenging.  Not sure what would help.  Maybe your PSU is overly sensitive to brownout?

zurdoid123, 2024-03-13:
I see, I will investigate. Thanks. 

The controller is powered by usb though. Also tried using a seperste psu for it but didnt notice any difference.