terjeio/ioSender #366: Chinese translation

hanke-cnc, 2024-03-14:
[中文已翻译.zip](https://github.com/terjeio/ioSender/files/14598764/default.zip)

terjeio, 2024-03-25:
FYI adding this will take a bit of time since the translation is using files that I do not have an [established workflow](https://github.com/terjeio/ioSender/wiki/Localization#the-iosender-way-preferred) for.

hanke-cnc, 2024-03-26:
> FYI adding this will take a bit of time since the translation is using files that I do not have an [established workflow](https://github.com/terjeio/ioSender/wiki/Localization#the-iosender-way-preferred) for.

Are you suggesting that we should edit the translation.txt by using Excel, instead of translating the CNC.Controls.WPF.resources.en-US.csv?

hanke-cnc, 2024-03-26:
> FYI adding this will take a bit of time since the translation is using files that I do not have an [established workflow](https://github.com/terjeio/ioSender/wiki/Localization#the-iosender-way-preferred) for.

I may have misunderstood. I saw that the translation.txt was disorganized and I didn't think to open it with Excel. I thought you would automatically generate the final translation.txt from the CNC.Controls.WPF.resources.en-US.csv file.

hanke-cnc, 2024-03-26:
> FYI adding this will take a bit of time since the translation is using files that I do not have an [established workflow](https://github.com/terjeio/ioSender/wiki/Localization#the-iosender-way-preferred) for.

"If I have free time tomorrow, I will re-translate and submit from translation.txt."

closed 2024-03-26 by hanke-cnc

terjeio, 2024-03-26:
I've managed to convert your files to the translation.txt format. Looks ok to me, do you want me to add it?

[translation.zip](https://github.com/terjeio/ioSender/files/14756741/translation.zip)

hanke-cnc, 2024-03-26:
> I've managed to convert your files to the translation.txt format. Looks ok to me, do you want me to add it?
> 
> [translation.zip](https://github.com/terjeio/ioSender/files/14756741/translation.zip)

yes

hanke-cnc, 2024-03-26:
> I've managed to convert your files to the translation.txt format. Looks ok to me, do you want me to add it?
> 
> [translation.zip](https://github.com/terjeio/ioSender/files/14756741/translation.zip)

No, wait a minute, I will make some translation corrections based on this translation.zip later.

hanke-cnc, 2024-03-26:
> translation.zip

Made some minor translation corrections to translation.zip

[translation_V2.zip](https://github.com/terjeio/ioSender/files/14758723/translation_V2.zip)

hanke-cnc, 2024-03-26:
> I've managed to convert your files to the translation.txt format. Looks ok to me, do you want me to add it?
> 
> [translation.zip](https://github.com/terjeio/ioSender/files/14756741/translation.zip)
[translation_V2.zip](https://github.com/terjeio/ioSender/files/14758733/translation_V2.zip)

Made some minor translation corrections to translation.zip

hanke-cnc, 2024-03-27:
> Translated [edge versions](http://io-engineering.com/downloads) now uploaded - 2.0.45p5.

This website of yours cannot be opened? HTTP ERROR 503

terjeio, 2024-03-27:
It seems my provider has changed the site to https only, so try [this link](https://www.io-engineering.com/downloads/).

hanke-cnc, 2024-03-27:
> It seems my provider has changed the site to https only, so try [this link](https://www.io-engineering.com/downloads/).

It's great, but the tabs on the right such as jog are always open and embedded in the main interface, instead of opening in another sub-window in the old version. Is this the latest change?

[image]

terjeio, 2024-03-27:
That is the XL variant, try the non XL one.