terjeio/ioSender #367: Problem with probing

Varra, 2024-03-26:
In my G-Code file I have several M66P0L4Q5.000 commands and it works as expected. But after applying height map this command disappear everywhere. Even my custom G-Code M101 is remain but M66P0L4Q5.000 is disappear. After applying height map editing button is inactive and it's not possible to edit file and add these commands manually.
Another problem when probing with "Set Z = 0 at X0Y0:" option and negative X Y in "Area to probe". Z is tot become 0 at X0Y0 in .map file. For example: Area to probe X -150, Y -150, W 300, H 300,  Grid 50mm.  Z0 will be at X-150 Y-150 but not at X0 Y0.
Or may be I am not understand how this option should work.

It also would be nice to add possibility to edit and save G-Code file after applying height map.

terjeio, 2024-03-26:
New [edge versions](http://io-engineering.com/downloads) with M66 bugfix uploaded - 2.0.45p5.

> Another problem when probing with "Set Z = 0 at X0Y0:" option and negative X Y in "Area to probe". Z is not become 0 at X0Y0 in .map file. 

Which version are you running?

> It's also would be nice to add possibility to edit and save G-Code file after applying height map.

Save then edit works.

> My custom code visible but doesn't work after applying height map either. M101L1Q60 should send command via UART but it doesn't.

I'll have to look into how to better handle arbitrary M-commands when transforming the gcode, e.g. the L word is removed in the above example.

terjeio, 2024-03-31:
>  the needle touches and scratches the surface in many places.

What kind of motion? Rapids (G0) or regular (G1, G2 or G3)?

Are the map files above generated from scanning the same surface? And by following the same pattern?

Varra, 2024-03-31:
Motion is G1, normal.
Patterns are exactly same. 300x300mm, X-150 to 150, Y -150 to 150, grid size 50mm. Surface is same.
Only difference measuring direction. ioSender starts from X-150 Y-150  and than moves by Y axis (Y-100, Y-50, Y 0, etc.) and Candle starts also from X-150 Y-150 but than moves by X axis   (X-100, X-50, X 0, etc.)

Varra, 2024-03-31:
[image]

Varra, 2024-03-31:
[image]

terjeio, 2024-04-08:
Here is the hegthmap from Candle, offset to 0 for the origin.

```
0;-0,082;-0,136;-0,162;-0,137;-0,087;-0,037
-0,037;-0,109;-0,16;-0,187;-0,182;-0,129;-0,051
-0,013;-0,087;-0,141;-0,173;-0,172;-0,116;-0,047
-0,03;-0,102;-0,154;-0,176;-0,179;-0,13;-0,057
-0,071;-0,144;-0,195;-0,222;-0,224;-0,169;-0,106
-0,128;-0,199;-0,252;-0,276;-0,276;-0,222;-0,15
-0,133;-0,242;-0,302;-0,333;-0,336;-0,279;-0,215
```
As you can se there is quite a bit of difference between them suggesting to me that the probed surface might be bending when probed.  This might be due the probing speed beeing different. Or a bug somehwere causing the heighmaps to differ - it looks like ioSender is not probing the full width of the surface.
The differences are largest on the right hand side, is this where the scraping occurs?

Varra, 2024-04-12:
Yes, left and right side are different height after applying a height map. I've changed a probe and made more tests. Circle 300mm. On the photo you can see difference.
  
[image]

Varra, 2024-04-12:
Another strange behavior. I put probe to Z 1mm and start probing. And after probing is done probe returns to approx the same position but ioSender shows Z -4.xxx mm which is wrong. Not big problem because anyway I have to set new zero for Z axis with tool. But I think it should not be like this.

terjeio, 2024-04-12:
Is Candle probing each point twice with different feed rates like ioSender does?
Any backlash in the Z-axis?

Varra, 2024-04-12:
No, Candle probing only once, but speed I use is same - 10 mm/min.  And Candle shows immediately result of each point and visualization during probing ))
I think there is very slight backslash on Z axis. But it's not so big and result I still get acceptable for my need.
[image]

Varra, 2024-04-12:
I would also like to thank you for this amazing project. Helped me a lot...

Varra, 2024-04-12:
BTW. Now I'm using ESP32 TinyBee board with grblHAL and would like to ask you does any ESP32 board support I2C keypad? I've tried but no luck. I2C EEPROM is working but keypad is not. Now using keypad with UART. Not sure is it right place to ask...

terjeio, 2024-04-12:
> ioSender shows Z -4.xxx mm which is wrong.

Do you have _Set Z = 0 at X0Y0_ checked when probing?

> No, Candle probing only once

Ok, then you may set the ioSender _Latch distance_ to 0 to skip the slow step. If you do so are the maps more equal?

>  does any ESP32 board support I2C keypad?

[This board](https://github.com/grblHAL/ESP32/blob/892a2bdbbf513b4f0e74c20701d96821a733563b/main/boards/cnc_boosterpack_map.h#L115-L117) has the pin definition for the keypad strobe.

Varra, 2024-04-16:
> Ok, then you may set the ioSender _Latch distance_ to 0 to skip the slow step. If you do so are the maps more equal?

Yes. Now much better!!! Thank you!!!  I also hope one day ioSender will stop braking my custom G-Code )))

> [This board]  has the pin definition for the keypad strobe.

Ok. I will try.  Is there definition example to use UART 3 in ESP32 boards? If I want to use MPG on UART 2 and UART keypad (input) and Debug (output) on UART 3. Is it possible? Couldn't get it to work.

terjeio, 2024-04-16:
> I also hope one day ioSender will stop braking my custom G-Code )))

This is a bit complicated since a block (line) may contain several gcodes and some parameter words are single meaning. If the gcode is unknown (which parameter letters are asscociated with it) it becomes impossible to parse unambiguously.
I _could_ assume that an unknown gcode "owns" all the parameter letters and just keep them (and the values) in a string and pass them unchanged through the transform routines and I may do this later.

> Is there definition example to use UART 3 in ESP32 boards?

Not yet? You have to use instance number 2 to claim it - but it cannot be claimend by two "clients" unless you modify the code by commenting out or removing [this line](https://github.com/grblHAL/ESP32/blob/383f7b81da8cd223fe5350db396a6cd3566f6105/main/uart_serial.c#L1061).

Varra, 2024-04-19:
> I _could_ assume that an unknown gcode ...

Is it necessary to apply height map to M-Codes? Why not parse only G-code and leave M-codes as they are?

> Not yet? You have to use instance number 2 ...

Thank you so much!!! With your help everything works now. I2C, Serial2, Serial3...
In the file "uart_serial.h" line 132 there is missing "," after "}" but error come out only if  "SERIAL3_ENABLE" is defined.
I guess no one else accept me needed Serial3 )))

Keypad plugin also contain display plugin but there is no description. Is there any info or examples? Will it work with ESP32 boards?
I tried to compile with this plugin enabled but got errors:
"msg_type_t too large for I2C display interface"
"machine_state_t too large for I2C display interface"
"coord_system_id_t too large for I2C display interface"

terjeio, 2024-04-19:
> Why not parse only G-code and leave M-codes as they are?

I'll have to leave _unknown_ M- and G-codes alone, and that can be tricky as explained above.
BTW gcode transforms other than the heightmap transform also uses the parsed/tokenized gcode, and I am not going to write a separate parser just for the heightmap. So give me some time.

> In the file "uart_serial.h" line 132 there is missing ","

I'll fix this in the next commit.

> Keypad plugin also contain display plugin but there is no description. Is there any info or examples?

It is for the [Jog2K Jog Controller](https://github.com/Expatria-Technologies/RT_Jog_Controller). Or for MPGs or displays that uses the [grblHAL message parser](https://github.com/grblHAL/grblHAL_MessageParser).

> Will it work with ESP32 boards?

No - at least not yet. The ESP compiler do not understand the assert statements and currrently the I2C driver code does not support the required API functions. So a bit more code, and testing, is required to make it work.