terjeio/ioSender #368: STM32 Blackpill?

BetaAsPhuck, 2024-03-28:
Does IoSender Work with STM32 Blackpill? (STM32F401CCU6)

IoSender doesn't recognize it when I connect the board to my computer via USB.

My computer recognizes the STM when I run it as BOOTLOADER, but doesn't recognize it in the device manager otherwise.

Also my first time working with stm32 blackpill, so any mistake is possible.

The cable works though, since I've uploaded code using it.

I **uploaded uCNC Firmware to it**

Any Help?
[image]
[image]

Arduino UNO is recognized for reference even though it's not running any relevant firmware

terjeio, 2024-03-28:
It should work provided that the board runs Grbl compatible code and enables a USB CDC stack for communication.

You may try [grblHAL](http://svn.io-engineering.com:8080/?driver=STM32F4xx&board=BlackPill%20%28STM32F401%29) - if that works then uCNC does not enable a USB CDC stack or fails to do so.

BetaAsPhuck, 2024-03-29:
I used a USB to USB C cable to program the board, by "DFU". I haven't used the Serial cable , I'll get one today and try that.

Once I've uploaded the code and give power externally to the board (or still via the USB-C), the Diode doesn't blink every second.
The Power LED shines, but the PC13 Builtin LED Doesn't.

Update:

OK So here's the thing, I tested uploading just the basic LED code onto the chip to see if it works, and it didn't.

I Erased All memory from the chip using STLink Utility and STLink V2 Board. And then the LED Sketch worked, I tried uploading the uCNC Code right after. Didn't light up again.

I Erased All memory again, uploaded uCNC to the Clean memory and then everything worked perfectly.

This may be a normal STM32 thing or a counterfeit thing that I just wasn't aware of. (Also the board came locked in RDP_1 for some reason so definitely a sketchy board with all these things combined)

Conclusion:
Reuploading isn't a thing on this board, only uploading the code after erasing all memory makes the code actually run.

Anyway, my problems are resolved, the board is recognized via my USB-C Cable, and works with ioSender.

phil-barrett, 2024-03-29:
Does your "Blackpill" have a real STM chip on it?  Many of the STM32 parts have been cloned and unscrupulous vendors use remarked clones to get a better price. The STM32F1xx line has at least 8 known knockoffs. The GD line is a pretty good clone, others are not so good, though. I believe the F4xx series has a couple. It's one of the reasons I did not use STM parts for my BOBs.  You just never know what you are going to get.

terjeio, 2024-03-30:
The chip is either mislabeled after beeing recycled or a clone since the [device id does not match the labeling](https://github.com/Paciente8159/uCNC/issues/671#issuecomment-2025564466), also the ST logo is missing.

closed 2024-03-30 by terjeio