terjeio/ioSender #370: Lathe taper problem.

nickshl, 2024-04-06:
Hi @terjeio, could you try to cut taper? But with real part please :)

There a problem in this line: https://github.com/terjeio/ioSender/blob/614115d80c1f6c792112371e94f73741558cc407/CNC%20Controls%20Lathe/CNC%20Controls%20Lathe/TurningLogic.cs#L181C1-L181C100

It should be like this(**see the comment**):
```
if (angle != 0.0d)
{
    ztarget = cut.Distance / angle * model.config.ZDirection;
    model.gCode.Add(string.Format("G1 X{0} Z{1}", model.FormatValue(diameter), model.FormatValue(zstart + ztarget)));
}
else
{
    model.gCode.Add(string.Format("G1 Z{0} F{1}", model.FormatValue(ztarget), model.FormatValue(feedrate)));
    model.gCode.Add(string.Format("G0 X{0}", model.FormatValue(xtarget + xclearance))); // THIS LINE MUST BE INSIDE else STATEMENT !!!
}
model.gCode.Add(string.Format("G0 Z{0}", model.FormatValue(zstart + model.config.ZClearance / model.UnitFactor)));
```
I almost ruined the workpiece because of that. I actually ruined it anyway, but it completely my fault.

nickshl, 2024-04-17:
@terjeio any updates on this issue? This is an example with 1 mm cut, clearance, last pass 45 degree taper:

> G18 G8 G21
> M3S100 G4P1
> G0 X6.000
> G0 Z0.500
> G97
> (Pass: 1, DOC: 4.000 1.000)
> G1 X4.000 F200.000
> G1 X5.000 Z-1.000
> **G0 X5.000**; // THIS LINE SHOULDN'T BE THERE !
> G0 Z0.500
> (Pass: 2, DOC: 3.000 1.000)
> G1 X3.000 F200.000
> G1 X5.000 Z-2.000
> **G0 X4.000**; // SO THIS!
> G0 Z0.500
> (Pass: 3, DOC: 2.000 1.000)
> G1 X2.000 F200.000
> G1 X5.000 Z-3.000
> **G0 X3.000**; // AND THIS!
> G0 Z0.500
> (Pass: 4, DOC: 1.000 1.000)
> G1 X1.000 F200.000
> G1 X5.000 Z-4.000
> **G0 X2.000**; // THIS ONE TOO!
> G0 Z0.500
> (Pass: 5, DOC: 0.000 1.000)
> G1 X0.000 F100.000
> G1 X5.000 Z-5.000
> **G0 X1.000**; // At this point it will be 2 mm rod instead taper...
> G0 Z0.500

Without fix, it cut taper, then at the end of this taper it moves cutter to the center. 
This is a bug to use taper minor diameter to calculate clearance, taper **major** diameter should be used. Or it shouldn't be used at all since it unnecessary - when we move Z to start position, it immediately clear from part because of taper.

terjeio, 2024-04-17:
Not yet, I have been quite busy with private assignments lately.

terjeio, 2024-04-21:
I have uploaded new [edge versions](https://www.io-engineering.com/downloads) for you to try - when a taper is selected only current and target diameter is used for the calculations - and Z length is calculated as well. I have also enabled entry of fractional degrees and I flag an error if the angle is >= 90 degrees. I have not yet implemented/verified internal tapers and the UI could be improved.

What do you think?

nickshl, 2024-04-22:
Looks good now. How it used to be, you can check there: https://youtu.be/5Wq_Iub6t_M?t=272

closed 2024-04-25 by terjeio