terjeio/ioSender #373: Machine moving unexpectedly when on probing page

imechura, 2024-04-09:
Hi I have just started using the probing tab on ioSender. In the past I have used the probe by manually entering gcode and it worked but when I change to the probing page, the machine automatically enters a probing sequence without me hitting the start button. If stop it before it crashes and then use the start button to run an edge finer or center finder sequence then the machine will repeatedly run the probing sequence over and over again sometimes running the machine into its limit switches. I could not find a manual that explained the expected behavior for the probing page. Is what I described what should be expected?

please advise.

imechura, 2024-04-10:
I’m trying to upload a video but this site is limited to 10mb for a video upload?

terjeio, 2024-04-10:
Try the latest [edge version](https://www.io-engineering.com/downloads). Ok?

imechura, 2024-04-11:
Ok, I will try it tomorrow

terjeio, 2024-04-11:
You may also try to short the cycle start input to ground. If EMI somehow triggers the input it may cause the observed behaviour.

imechura, 2024-04-19:
It was emf on the floating cycle start pin.

closed 2024-04-19 by imechura