terjeio/ioSender #378: Unexpected results with G38.2

imechura, 2024-04-19:
I have two probing issues that might bor might not be related:

First:

Whenever I issue a G38.2 command the machine does move to the stated location.

for instance:
G0 X0
G91 G38.2 F300 X10  // This should move X 10 mm  but instead it moves what appears to be a random distance.

If I issue the same command repeatedly it will move a different distance for each issuance.

However, if I use the following command:
G0 X0
G91 G01  F300 X10 // 

The G01 command will move the table exactly 10mm every  time I run it.

Second:

When using the probing page for external center finder, It probes the edges in the wrong sequence and it moves the prob in the wrong direction,

terjeio, 2024-04-19:
> Whenever I issue a G38.2 command the machine does move to the stated location.

EMI is triggering the probe input?
Here is what I get in the console by repeating `G91 G38.2 F300 X10` a few times:
```
G91 G38.2 F300 X10
ALARM:5
[MSG:Stop]
[PRB:10.000,0.000,0.000:0]
G91 G38.2 F300 X10
ALARM:5
[MSG:Stop]
[PRB:20.000,0.000,0.000:0]
G91 G38.2 F300 X10
ALARM:5
[MSG:Stop]
[PRB:30.000,0.000,0.000:0]
G91 G38.2 F300 X10
ALARM:5
[MSG:Stop]
[PRB:40.000,0.000,0.000:0]
G91 G38.2 F300 X10
ALARM:5
[MSG:Stop]
[PRB:50.000,0.000,0.000:0]
G91 G38.2 F300 X10
ALARM:5
[MSG:Stop]
[PRB:60.000,0.000,0.000:0]
```
You get?

> When using the probing page for external center finder, It probes the edges in the wrong sequence and it moves the prob in the wrong direction,

Works for me, probing directions are +X, -X, +Y, -Y as indicated in the graphics. What is your sequence?
Console log, I am probing a hex nut with the spindle loose so accuracy is not that good:
```
[PRB:29.640,112.288,-29.669:1]
[PRB:29.715,112.288,-29.669:1]
[PRB:56.943,112.288,-29.669:1]
[PRB:56.853,112.288,-29.669:1]
[PRB:43.305,99.390,-29.669:1]
[PRB:43.305,99.425,-29.669:1]
[PRB:43.305,125.185,-29.669:1]
[PRB:43.305,125.163,-29.669:1]
```
A more comprehensive log is generated if you check the "Verbose" tick box.

imechura, 2024-04-19:
Here are videos that show the behavior. I will get the rest shortly.

https://photos.app.goo.gl/srPLQSZppKc2vQh29

imechura, 2024-04-19:
The sequence is 1, 2, 4, 3 based on the labels in the graphic.

I am going to look into EMI now (again) that would explain the randomness that I am seeing. That probe is powered by a shielded usb-c cable so in theory it should be ok but in practice who knows.

phil-barrett, 2024-04-19:
What breakout board are you using? The fact that it is random surely implies EMI.  Shielded USB has nothing to do with the probe input.  All inputs, including probe, should have some sort of isolation or, at the very least, low pass filtering. 

You might want to look [this](https://www.grbl.org/single-post/emi-science-voodoo-or-doodoo) over.  It gives a fairly broad overview of EMI and points you in the direction of various things to do about it.

imechura, 2024-04-27:
I have spent countless hours testing for EMI and I have replaced all of the probing wiring with correctly shielded cables and connectors and the issues still persists. However the issue is completley resolved by reverting to ioSender v2.0.42p6. That leads me to beleive that it not caused by interference or an issue with my board or settings. I guess I will just stay on this older version for a while.

terjeio, 2024-04-27:
The external center finder runs in the correct sequence in v2.0.42p6?
I asked what responses you get in the console when probing in [this comment](https://github.com/terjeio/ioSender/issues/378#issuecomment-2065795357) but you did not answer, the best would be with _Verbose_ checked in the console. I need this so that I can get a clue about what is going on on your end - this since I am not able to replicate the behaviour on my end.

closed 2025-02-07 by terjeio