terjeio/ioSender #382: Shortcut key issues

auxp, 2024-05-04:
Can keyboard shortcuts move the Z-axis or A-axis

terjeio, 2024-05-05:
[Yes](https://github.com/terjeio/ioSender/wiki/Setup-and-configuration).
A-axis is controlled by \<ctrl\>+\<shift\>+U and N, Z by  \<ctrl\>+\<shift\>+I and M or \<PgUp\> and \<PgDn\>.
Save out a keymapping file from the _Settings: App_ tab and modify it if you want to change any mappings.

auxp, 2024-05-14:
> [是的](https://github.com/terjeio/ioSender/wiki/Setup-and-configuration)。 A 轴由 <ctrl>+<shift>+U 和 N 控制，Z 由 <ctrl>+<shift>+ I 和 M 或 <PgUp> 和 <PgDn> 控制。从_“设置：应用_”选项卡中保存键盘映射文件，如果要更改任何映射，请对其进行修改。

My computer has two keyboards, why is it that one of them cannot achieve multi key functions such as stopping, starting, resetting, unlocking, and resetting, while the A-axis can be tapped

terjeio, 2024-05-14:
Not sure I understand the question...
Anyway, see this discussion for how you can [log kepresses to the console](https://github.com/terjeio/ioSender/discussions/295), perhaps that will help?