terjeio/ioSender #383: WEIRD things arre happening

smitroast, 2024-05-12:
just recently built a big enclosure for my 700x700 work space cnc....

so i extended my wires made new wires etc (everything worked perfectly before hand)

when i first turned it back on all jogging operations worked normally

then homing operations stated playing

Z and X axis both homing at the same time no matter what settings i had selected and then Y axis would home hit limit switch and stop no second hit and homed, just one hit and stopped...

tried changing wiring ETC

anyway now my IOsender is locked out with an ALARM 10

i dont have an e stop

ive tried everything pulled all the limit switches out of the controller

fully reset and re downloaded windows on my computer and nothing works

i use a makers E5XMCS T1.4

terjeio, 2024-05-12:
Start with [First Run Settings](https://github.com/grblHAL/core/wiki/First-Run-Grbl-Settings).

smitroast, 2024-05-13:
> Start with [First Run Settings](https://github.com/grblHAL/core/wiki/First-Run-Grbl-Settings).

have done this now i still have alarm 10 with an error 13 but i have not door limit or anything attached to the port

smitroast, 2024-05-13:
i ended up removing all errors but

y axis is the only one that moved by jogging

HOMING cause Z and X to move at the same time

so i did $rst=* and now im stuck again having ALARM10 popping up

and a pop up on the screen asking to shut down IOsender because Estop is still active

smitroast, 2024-05-13:
sorry another update

unlocked it again

jogging X in - dir does nothing

jogging Z in - does both Z and X

terjeio, 2024-05-13:
If it worked before rewiring and the firmware has not been changed then either the rewiring has errors (shorts?) or the controller has been damaged in some way?

smitroast, 2024-05-13:
Yeah I’m starting to think the controller has been damaged somehow

terjeio, 2024-05-13:
Well, it is odd that motion works in one direction but not in the other - is the step signal lost due to the direction signal shorting it? Or is the enable signal in the wrong state (enable and dir swapped)?

smitroast, 2024-05-13:
I will have to have a look tomorrow or later tonight I’m sure everything is in the same spot as it was in before. It’s just weird that it doesn’t job but moves when homing but does Z and X at the same time