terjeio/ioSender #386: Using a tool offset for probing offcenter tools

T4KUUY4, 2024-06-11:
Hi, 
Im unable to find out if its possible to use tool diameter offsets for toolchanging touch off´s so you can automatically touch off face mills etc.

Tools that are "deeper" in the center like face mills. 

Is this possible in IoSender?

terjeio, 2024-06-17:
Is this in combination with touch off @ G59.3?
FYI the toolchange workflow is run on the controller side, the sender is basically waiting for a cycle start to be issued (while allowing jog commands).

T4KUUY4, 2024-06-18:
Hi, 
yea it is in combination with the automatic touch off at G59.3.

When using a normal size endmill, everything works out great. I hit M6 T5 for example, toolhead raises up, i change tool, click start and it moves over to the pin + touches off. All good for normal style endmills. 
But facemills for example, are deeper in the center than on the outside (where the inserts are), so the measurement is not right. 

These would require a different touch off position so the inserts touch the probe and not the center depper dish of the facemill. 
See examples below:
[image]
[image]

Thanks in advance,
Tom

terjeio, 2024-06-19:
In the next commit I will change the signature of an event and where it is called from to allow plugin code to modify the G59.3 target. A simple plugin will be something like this:
```
#include "grbl/hal.h"
#include "grbl/protocol.h"

static on_probe_toolsetter_ptr on_probe_toolsetter;
static on_report_options_ptr on_report_options;

bool probeToolSetter (tool_data_t *tool, coord_data_t *position, bool at_g59_3, bool on)
{
    if(on && tool && position) {

        if(tool->tool_id == 5)
            position->y += 5;
    }

    if(on_probe_toolsetter)
        on_probe_toolsetter(tool, position, at_g59_3, on);

    return true;
}

static void report_options (bool newopt)
{
    on_report_options(newopt);

    if(!newopt)
        hal.stream.write("[PLUGIN:Toolsetter tool offset v0.01]" ASCII_EOL);
}

void my_plugin_init (void)
{
    on_probe_toolsetter = grbl.on_probe_toolsetter;
    grbl.on_probe_toolsetter = probeToolSetter;

    on_report_options = grbl.on_report_options;
    grbl.on_report_options = report_options;
}
```

A more complete plugin should be made that allows setting the offset(s) for multiple tools and store them in EEPROM. And maybe later incorporate the code into the core itself.

T4KUUY4, 2024-06-19:
This sounds great. Cant wait for it to happen.
Thanks a bunch

terjeio, 2024-07-08:
Here is a plugin for you to test:
[my_plugin.zip](https://github.com/user-attachments/files/16134070/my_plugin.zip)

It has five slots (compile time configurable) for specifying tool offset (radius), applied in the X direction (should be configurable via setting?). Configuring a slot is via M102: `M102 P<slot> T<tool> R<tool radius>`

I do not know if this plugin cover the needs for probing odd tools, feedback is required before I add it as an option in the Web Builder.

T4KUUY4, 2024-07-09:
Thanks, Ill give it a spin soon and then ill let you know if any issues come up.

m-basov, 2024-07-23:
Hi @terjeio! I tested your plugin with some minor changes to Grbl and it seems to work well! I tested it on Flexihal https://github.com/Expatria-Technologies/STM32F4xx/compare/F446_Flexi_HAL...m-basov:STM32F4xx:tool_offset_plugin#diff-ea19d2ee44f9bcf77cfc7473d38a797ee5839d6f5791b005a9de0a3e73d5a295R166 

https://youtu.be/89Yvjancsg8

GameOEver, 2024-09-07:
@terjeio I used the version of @m-basov but it looks like the offset is applied to Y? Unfortunately a movement in Y doesn't change the location of my endmill nor the toolsetter. Would you please guide me to where I can change that offset to apply to X?

Thanks in advance!

Best regards

GameOEver, 2024-09-07:
+ I didn't find a setting for that in IOsender after upgrading the firmware

GameOEver, 2024-09-07:
nevermind, @m-basov  chaged it to apply to Y instead of X

closed 2025-02-07 by terjeio