terjeio/ioSender #392: X gantry squaring process, perhaps something for a wizard in IOSender?

5ocworkshop, 2024-06-30:
Hi Terje,

I'm afraid it has been almost a year since my move and I am still trying to find workshop space so my tools are in storage and I am still between machines.  I see the project has continued to grow in leaps and bounds.  Nice work.

I wanted to share a short PDF I prepared back at my old workshop that details how to use a probe and a corner block to square the X gantry to the Y rails on a CNC machine using math and the machine itself to measure the deviation.  I realize there are multiple ways to get close to square but this is deterministic and easily repeated.  And it should result in virtually perfect square (it did for me and others who have tried it).  The docs are for grblHAL and IOSender.  I added it to github here: https://github.com/5ocworkshop/squarecnc

Perhaps an idea for a future wizard in IOSender?

phil-barrett, 2024-06-30:
Nice. Using probing to help square the machine is a great idea.

One thing, when you cut the pocket with an unsquared machine, the pocket will be a parallelogram. Probably close to a rectangle. Inserting the probe base into the pocket will likely not have the Y Sides of the probe precisely parallel to the Y Axis.  So, you may need to oversize the pocket and make sure that the Y axis is  pressed firmly against a Y side of the pocket.  

I think the distance between the screws (your triangle "height") should actually be the distance between the sensor contact points.  The sensor targets on my Avid are about 10 mm from the rack (no Y screws, rack and pinion) so measuring between the rails would be off by 20mm or 20/1250 (1.6%).

Also, a small typo - you wrote the distance between your Y rails is 1248 and in the calculator you used 1428.

phil-barrett, 2024-06-30:
On a wizard (or probe panel).  I like the idea.  It should also support 3D touch probes.

5ocworkshop, 2024-07-02:
> Nice. Using probing to help square the machine is a great idea.

Thanks.  It always struck me as strange that we have these high precision measuring machines and then people use things like jogging and dropping a pin in a hole to try and solve this issue.

> One thing, when you cut the pocket with an unsquared machine, the pocket will be a parallelogram. Probably close to a rectangle. Inserting the probe base into the pocket will likely not have the Y [...]

It has been quite a while since I did it (I'm without a machine at the moment) but I used MDF on purpose to ensure I have a little bit of give on the insertion.  As I recall it went in fine but was difficult to get out.  I took the MDF off the machine and pressed it through IIRC.

> I think the distance between the screws (your triangle "height") should actually be the distance between the sensor contact points. The sensor targets on my Avid are about 10 mm from the rack (no Y screws, rack and pinion) so measuring between the rails would be off by 20mm or 20/1250 (1.6%).

I've been thinking about this since I saw your comment.  In relation to a ball screw driven machine, the carriages on the Y axis are centered on the ball screw and represent the mechanical anchor points of the X gantry so using that centerline will result in the adjustment to get, conceptually, to perfect mechanical square.  (my typo that you note below aside, thanks for catching that.)
 
I'm not familiar enough with the mechanics of your rack and pinion driven machine, but I would think you still ultimately want the centerline of the mechanics of the axis for use in the calculation.

> Also, a small typo - you wrote the distance between your Y rails is 1248 and in the calculator you used 1428.

I'll see if I can find the original word doc and fix that.  The larger value is the correct value (the one I used in the calculation) for my particular machine.  My machine is a bit odd, it has more X and Y travel as it was a custom build. 
 
> [edit] By the way, you could also use a 1-2-3 Block instead of a fancy probe. That is more likely to be square.

You are correct, a 1-2-3 block would be the typical thing to use (I thought I mentioned that in the doc?) but I don't have one, and the corner probe I had was custom made for me on a professional mill so I was pretty comfortable with the squareness and finish.

I hope this helps someone new setting up a machine or calibrating their machine.  The fact it is deterministic and returns measured values really appealed to me.

phil-barrett, 2024-07-02:
R&P vs Ball Screw doesn't matter.  It is the measurement points of the sensors since that is what grblHAL is offsetting from.  The sensors form one leg of the triangle, not the anchor points.