terjeio/ioSender #394: 2.0.25p8 probbing issue

calabr, 2024-07-15:
Get issue with TLO probing 
IoSender 2.0.25p8,  classic GRBL 1.1 controller
After successful probing cycle completed, G54 Z offset become 0.
Z axis rapidly moved top Z position, tan back probing  start position.  
  The same behavior verified with 2 different PC.

CLI log:
`Probing...
G91F50
PM:ok
G38.2F50Z-10
[PRB:-259.315,-157.635,-69.270:1]
PM:ok
G0Z0.5
PM:ok
G38.2F25Z-2
[PRB:-259.315,-157.635,-69.275:1]
PM:ok
G53G0Z-69.275
G53G0Z0
G10L20P1Z0
G53G0Z-67.053
[GC:G0 G54 G17 G21 G91 G94 M5 M9 M56 T0 F25 S0]
[G54:-263.083,-152.097,0.000]
[G55:-136.618,-93.240,-4.500]
[G56:-140.164,-148.264,-42.895]
[G57:0.000,0.000,0.000]
[G58:-100.000,-100.000,-60.000]
[G59:-144.000,-91.000,-82.800]
[G28:0.000,0.000,0.000]
[G30:0.000,0.000,0.000]
[G92:0.000,0.000,0.000]
[TLO:0.000]
[PRB:-259.315,-157.635,-69.275:1]
G90
Probing completed.
`
Probing settings
[image]

phil-barrett, 2024-07-15:
2.0.25 is pretty old.  Perhaps you should update to the current version? 2.0.44 is the latest.

calabr, 2024-07-15:
> 2.0.25 is pretty old. Perhaps you should update to the current version? 2.0.44 is the latest.

 It's mistyping  issue with latest edge version  IoSender 2.0.45p8

terjeio, 2024-07-15:
This is an odd one that I do not understand, I copied the sources to my laptop a day before travelling and somehow a line has been added to the copy (the same day) that should not be there. I cannot remember doing it myself and there is no reason for adding it. I'll upload a new edge build when I have verified that there is no other unintended changes, likely tomorrow

calabr, 2024-07-15:
Thank you @terjeio !
Just now verified  2.0.24p6 - hath no this issue

terjeio, 2024-07-19:
Fixed in the latest edge build just uploaded.

closed 2025-02-02 by calabr