terjeio/ioSender #395: Whys is probing tab greyed out (disabled)?

stevempotter, 2024-07-16:
I have just updated to 2.0.44, I can't get to the Probing tab because it is greyed out. The machine is homed. Everything else seems to be fine.
It would be nice if there was a popup dialog when you click on something that is greyed out to tell you what needs to be done to enable that feature.
I have looked for answers by searching keywords here, and in the Wiki. Can't find any mention of a greyed-out probing tab.

terjeio, 2024-07-16:
$10 setting (_General > Status report options_) does not have _Probe coordinates_ checked?
It could also be due to an older grblHAL controller version.

stevempotter, 2024-07-31:
Yes, that was it, didnt have Probe Coords in status reports. Thanks!

Terje, I finally put my ESP-Now wireless pendant code up on github.