terjeio/ioSender #397: RE: Spindle

silver2row, 2024-07-27:
Hello...would I use `0 - 10v` or another section of the Spindle headers for a spindle that can handle 10,000 RPM?

Seth

P.S. T41U5XBB is the board I am using with the Teensy4.1. I was thinking that the Spindle needs a different set up since I did not populate the trimpot for the Spindle `0 - 10v` header. Please send guidance.

silver2row, 2024-07-27:
For instance, would I just use PWM and GND?

silver2row, 2024-07-27:
Or...if using PWM and GND only, would I need a driver or can I run the bare spindle wires directly from the T41U5XBB?

phil-barrett, 2024-07-27:
The trim pot is needed to set the 0-10V.  But, if your VFD supports 5V PWM, you can use that.  You will need to set up the VFD to take external input and 5V PWM. Connect to Gnd on the board to Gnd on the VFD (ACM or DCM, consult your VFD manual) You will also need to use a relay to connect FOR to ground (DCM, probably).

silver2row, 2024-07-27:
@phil-barrett ...thank you but I only have a cheap one for now. The two wires of the spindle are just that now, i.e. two wire interface.

The older controller plugged right in. I have no clue as to how it was wired, i.e. `not` open source stuff.

Seth

P.S. So, the 12v supposedly of this spindle can be directly attached to PWR of my PSU and then use GNDs all around? I will research more so I can understand a bit more.

silver2row, 2024-07-27:
Okay and A-Okay,

I was reading some. VFD or variable speed drives come in 110v and 220v. I will be purchasing one to handle 110v input while outputting some voltage and amperage. Please look here if you think my spindle would get along well with it:

https://filecenter.deltaww.com/Products/download/06/060101/Catalogue/DELTA_IA-MDS_VFD-EL_C_EN_20210420.pdf

That link is a .pdf file online from digikey's datasheets on their products they sell and handle. 

In the meantime, I can test the RPM with a `photo tachometer` to find out if indeed it is producing 7000 to 10000 RPM when the ioSender states it is using that speed.

Seth

P.S. Thank you first off for replying. Secondly, I appreciate you taking time out of your schedule to help out.

phil-barrett, 2024-07-27:
What spindle do you have?  Can you post a link to it?  Be careful because spindles come in 110 and 220 and need to be matched to the VFD for both voltage and power.

By the way, Delta VFDs are very good.  Pricey.  I've never used one but the manual is reputed to be very good.

silver2row, 2024-07-28:
Thank you for understanding so far. I usually mess with PWM lasers and other cheap machines so far. 

I am still workin' my way up the Totem pole in this onslaught on technology and electronics. I will confer to the manual. Bear with me.

Seth

silver2row, 2024-07-28:
@phil-barrett , hello again. I contacted the company that produced the machine. It is a small machine that does not handle a lot of torque or current. So, in hindsight, I should have broke down and got the machine that is better suited for my taste.

I just posted on their Q & A "forum" to find out about the spindle in question.

Seth

P.S. I will report back soon.

phil-barrett, 2024-07-28:
Could you please give me a link to the machine?

silver2row, 2024-07-29:
https://www.monoprice.com/product?p_id=44220

It is an inexpensive machine but I may get another set up soon. It seems the machine does not quite have the specifications one would need to handle nice parts. 

Seth

P.S. Thank you.

phil-barrett, 2024-07-29:
OK, that is a DC spindle.  You don't need a VFD.  It runs off 12 or 24V DC - not sure which, maybe either. Did the machine come with a controller?

silver2row, 2024-07-29:
Yes...the machine came with a controller.

phil-barrett, 2024-07-29:
Could take a photo of it?

silver2row, 2024-07-29:
Sure. Bear with me. I need to find it, dig it out of hibernation, and post the photo.

Seth

silver2row, 2024-08-01:
@phil-barrett ,

Do you have any news that may be helpful?

Seth

phil-barrett, 2024-08-01:
OK, looking at the board, it appears to be a woodpecker board.  That has a mosfet for driving a DC spindle or laser.  My boards don't have that.  To drive the spindle, you will need a separate driver for the spindle.  [Something like this](https://www.amazon.com/NOYITO-Voltage-Switching-Control-Transistor/dp/B07F5JPXYS/ref=sr_1_9?crid=2Y4612Q660QND&dib=eyJ2IjoiMSJ9.g8t6IGIeaAfpezNbzkabI24deRO7Q3FmrvtlJQZZR9QK7M3l7LsvNJ0Qusv7WHWVyO0iVFGZOL1WDKCwBXT4TPjok7HxQnl4orIs2Qn5VPPB3mPMIEVSMg1KqTQjd9bIjA_hfrHzXS1M_GNmuH5urjSHvtrG83jGU8QhVg36RLMXHW26aoPFMMfLFvXxmBSi5gLPOoCda8KQYjTIr32Lv2XJZAzQXDc6AB5B7XmUSoyQHAvTvlf-lBnJRUGDvT1zb6jhxF936lDHW4Iej9cBcSiFInx0jTx6xMw-748xvMM.-dBaQBTGklyl9ZnLJrWIxpnUQqc-synhFe9IGbV0ciw&dib_tag=se&keywords=mosfet+pwm+module&qid=1722531313&s=hi&sprefix=mosfet+pwm+module%2Ctools%2C134&sr=1-9) might work. I have ordered one and will give it a try.  I'll be out for a couple days. Hopefully I can try it out by monday.  I am still looking to see if there are better modules to use.

silver2row, 2024-08-01:
Thank you, sir. I may have a driver here I can test but I will need to check.

silver2row, 2024-08-05:
Also sir, I have been looking at M and G-codes for setting parameters or furthest extents. 

Is the GRBL g-code still 101 and 110 for XY? Off to research more. I keep getting other g and m-codes on different sites stating something different. I will go directly to GRBL.

Seth

silver2row, 2024-08-05:
https://www.diymachining.com/wp-content/uploads/2015/10/grbl-Settings-steps-per-mm-math-example.png

and...

https://www.diymachining.com/downloads/GRBL_Settings_Pocket_Guide_Rev_B.pdf

And A-Okay but...

1. Once I devise the math and set up the data to be understood by the controller and drivers, then what exactly?
    a. Can I set up my cartesian plane within your settings on the ioSender source?
    b. How would I set up the XYZ furthest extents to let my machine know not to go any further?

Seth

phil-barrett, 2024-08-06:
Here's a nice [grbl settings pocket guide](https://www.diymachining.com/downloads/GRBL_Settings_Pocket_Guide_Rev_B.pdf).  It doesn't have all the grblHAL ones but these are the basics.

As to GCode, I use the [LinuxCNC documentation](http://linuxcnc.org/docs/html/gcode/g-code.html).  It covers a lot more than grblHAL. 

As to setting max travel - $130, $131 and $132 will do that for X, Y and Z.  You will need to turn on soft limits.

silver2row, 2024-08-06:
Thank you sir. XY is cartesian...I thought. I am using Z also, i.e. so I guess just XY extents are not relevant. 

Seth

P.S. Thank you for the link. I will research it and thank you for making me understand the commands for max travel. I just did not type in `max travel` and was typing in google `furthest extents GRBL`. Oops. I got some odd results. Yes!

silver2row, 2024-08-07:
Okay. I found the max travel in the ioSender source...

I set up my Max Travel to X270 Y270 Z25. I have drivers that should handle the motors in question but I cannot get motion from my motors so far. 

I read in the Wiki you guys/gals put together that the app.config file can be setup to handle ideas also. 

I am not sure where to start so far...

At first my motors turned but not well. Now, the motors do not move at all. The motors are not hot like I would expect them to be hot if over powering them. I am using a 12v PSU with GeckoDrive G251 drivers and a cheaper Leadshine driver clone. 

Seth

P.S. At first, I think the reason is that I do not know how to wire the Leadshine driver clone to the T41U5XBB and Teensy4.1.

```
any help would be greatly appreciated...
```

Stuck at best right now...

phil-barrett, 2024-08-07:
You probably need to invert the stepper enable pin setting.  $4=7 will do it.

silver2row, 2024-08-07:
Okay. I will try it... Thank you.

silver2row, 2024-08-08:
@phil-barrett ,

Seth here once more. Do you think I should install the wires for the Disable Pins on the G251 drivers and the Enable Pin on the DM332T Leadshine Clone driver?

And then, do you think the inverting of the enable pin on the DM332T will do it?

Seth

P.S. I removed my laser project from the T41U5XBB, e.g. the machine was too cheap. The build was for a laser engraver/cutter but the lack of sturdy parts made the build irrelevant for now. Please send guidance. I can test the wiring but I am a bit mistook on what to do so far. Here is the G251 photo of connections and the DM332T connections:

[image]

and...

[image]

Also, if using the OPTO pin on the DM332T driver, which I think is needed, where should I attach it? GND or another header connector on the T41U5XBB?

phil-barrett, 2024-08-08:
OK, the DM332T has open collector interfaces.  What you will need to do is find a source of +5V.  If you don't have a +5V PSU, you can use the +5V terminal on the T41U5XBB.  Connect +5V to the opto input.  Connect the Dir, Step and Enable signals to DIR, PUL and ENA signals. No need to use Gnd if sourcing the +5V from the T41U5XBB.  You will need to invert Dir, Step and Enable in grblHAL settings ($2, $3 and $4) for the axes driven by the DM332T. 

For the axis driven by the G251,  you can connect DIRECTION, STEP, ENABLE and SIGNAL GND to the screw terminal pins on the T41U5XBB.  You may have to invert the enable signal for that Axis - try it both ways.

silver2row, 2024-08-08:
Thank you, sir...

I will do just that and get back to you. I really appreciate you making me understand possibilities with these drivers.

Seth

phil-barrett, 2024-08-08:
No problem.  Happy to help.

silver2row, 2024-08-08:
Hey sir...you were right. I got most of the drivers working XZ so far. Y is giving me issues. I will post a photo. If you see anything out of the ordinary, please let me know.

here:

[image]

Does the highlighted XYZ and other Red highlighted parts look okay?

Seth

silver2row, 2024-08-09:
> OK, looking at the board, it appears to be a woodpecker board. That has a mosfet for driving a DC spindle or laser. My boards don't have that. To drive the spindle, you will need a separate driver [...]

Sir,

Did you get around to the testing yet? Also, would I need a medium between the circuit and board or circuit and DC spindle?

Something like a controller to handle changing the speed of the device or do you think the source can handle that aspect of it all?

Seth

closed 2024-08-15 by silver2row