terjeio/ioSender #400: Attempts to run iosender with Genmitsu 3030 ProV Max

kgraven, 2024-08-14:
[image]
[image]

When attempting to start iosender (I tried 3 versions standard, XL, and the Edge one and all fail the same) I get Unhandled exception message shown in the screenshot above.  This is from a fresh config.  

I also tried to pull from github and build in Visual Studio (I downloaded the required old MS libs) and it does not build.  Is there a current one that builds out of the box so I can see where the code is failing?

Any help is appreciated.  Thanks!  
--Ken

terjeio, 2024-08-14:
Which controller/version?
It looks like something coming from the is not according to spec, post the `$I` response when connecting with a terminal please.

phil-barrett, 2024-08-14:
Genmitsu uses Grbl 1.1

kgraven, 2024-08-14:
And that is incompatible I'm assuming?

Sent from my Verizon, Samsung Galaxy smartphone

terjeio, 2024-08-15:
> Genmitsu uses Grbl 1.1

> And that is incompatible I'm assuming?

No, my guess is that it may be a custom version since the original I have on my Mega2560 works with the sender.

kgraven, 2024-08-15:
I get this in response to $I

[VER:ARM32 V3.0.20230816:CUSTOM]
[OPT:VZL,35,254]

Thanks,
Ken

Ken Gravenstede | L-Tron Corporation
Director of Engineering
P (585) 383-0050 x121   |   F (585) 383-0701 | C (585) 727-3691

7911 Lehigh Crossing, Suite 6, Victor, NY 14564

[cid:a71d9677-210e-4ba6-9ab5-af5e8aaba8f4]

Let’s Connect!  <https://twitter.com/LTronCorp> Twitter<https://twitter.com/LTronCorp> | <https://www.facebook.com/LTronCorp> Facebook<https://www.facebook.com/LTronCorp> | <https://www.linkedin.com/company/l-tron-corporation> LinkedIn<https://www.linkedin.com/company/l-tron-corporation> |  <https://plus.google.com/+L-tron/posts> Google+<https://plus.google.com/+L-tron/posts> | <https://www.youtube.com/user/LTRONCORP> YouTube<https://www.youtube.com/user/LTRONCORP> |

terjeio, 2024-08-15:
Found this [issue](https://github.com/winder/Universal-G-Code-Sender/issues/2524). Whoever made this fork added descriptions to the $$ settings values, which is not according to the v1.1 specification. I do not have the v3 specification and have not heard about it before - does it even exist?
Since the controller is ARM32 based and the binary I found is > 64K it might be possible to flash it with grblHAL depending on which processor is used. A link to the souce code would be helpful, but I cannot find it - perhaps it is not published (it should be)?

A bit more info in [these links](https://github.com/search?q=genmitsu&type=pullrequests).

phil-barrett, 2024-08-15:
I think it might be [this one](https://www.aliexpress.us/item/3256805760785240.html?gatewayAdapt=glo2usa4itemAdapt).

Not much info beyond Grbl 1.1h.  They claim Candle and Laser Grbl support.

wartech0, 2024-08-18:
The processor used in the 3030PROver Max is a GD32F103RCT6. I pulled it right out of a picture on my machine.
[image]

terjeio, 2024-08-24:
> The processor used in the 3030PROver Max is a GD32F103RCT6.

These processors are a bit hit and miss with grblHAL, some works - some not. And I have not seen this board before so IMO it is not worthwhile creating a board map for it. To run ioSender I suggest you replace the board with a supported one running an unmodified version of Grbl, or preferably grblHAL, as I am not too keen to change the sender to accomodate an undocumented Grbl variant I do not have access to.

kot0005, 2024-11-13:
any update on this issue ? I wanted to use iosender because of probing features. 

all the gcode senders work with this machine, So it shouldnt be too hard to to make it work with hobby machines.

terjeio, 2024-11-13:
More [here](https://github.com/terjeio/ioSender/issues/407#issuecomment-2474477006).

closed 2024-11-13 by terjeio