terjeio/ioSender #405: Enabling "Probe XY offsets" for Edge Finding and Height map

SebastianMusser, 2024-09-08:
Hi Terjeio,
as I added a retractable 3d probe to my printnc, I would love to use it for edgefinding etc. 

But in Iosender 2.0.44, the "**Probe XY Offset**" parameters are only availlable in the "**Center Finder**" tab... it is greyed out in EdgeFinding and Height Map etc... 

Is there a way to enable it? I would really appreciate it, otherwise I would have to go back to mounting the probe in my spindle...  

Thx!

[image]
[image]

terjeio, 2024-09-09:
It is available the edge finders if you select a probe action that involves XY motion. I'll have to add it to the Height map.

terjeio, 2025-01-28:
Initial code changes available the latest [edge version](https://github.com/terjeio/ioSender#edge-pre-releases). Creating the heigtmap should work, but applying it is another matter...