terjeio/ioSender #406: Lagging and reset?

risototh, 2024-09-12:
Not sure, what's happening there. I have implemented a M-code plugin into the grblhal for generic modbus communication (modbus-io). During development, I have tested it with in the UGS and/or console. Now, when I'm using it in the ioSender, there is some lag prior execution of the command and output of ```$G``` and ```$#```. Is the problem the output of the debug messages? I don't think so, as the messages are printed after the ```$G```/```$#```. This behaviour is not present in UGS or when directly connected via telnet.

I have executed the actual commands as macro:
```
M101D4E5P6Q1
G4P1
M101D4E5P6Q0
```

This  is the output in console:
```
<Idle|MPos:-6.000,-176.000,-57.005,0.000|Bf:100,1023|FS:0,0,0>
<Idle|MPos:-6.000,-176.000,-57.005,0.000|Bf:100,1023|FS:0,0,0>
;;; here is the macro executed (the button clicked)
[GC:G0 G54 G17 G21 G90 G94 G49 G98 G50 M5 M9 T0 F0 S0.]
<Idle|MPos:-6.000,-176.000,-57.005,0.000|Bf:100,1023|FS:0,0,0|H:1>
[G54:-645.000,-802.000,-145.500,-10.000]
[G55:-266.100,0.000,0.000,0.000]
[G56:0.000,0.000,0.000,0.000]
[G57:0.000,0.000,0.000,0.000]
[G58:0.000,0.000,0.000,0.000]
[G59:0.000,0.000,0.000,0.000]
[G59.1:0.000,0.000,0.000,0.000]
[G59.2:0.000,0.000,0.000,0.000]
[G59.3:0.000,0.000,0.000,0.000]
[G28:-2.000,-2.000,-2.000,0.000]
[G30:0.000,0.000,0.000,0.000]
[G92:0.000,0.000,0.000,0.000]
[T:1|0.000,0.000,0.000,0.000|0.000]
[T:2|0.000,0.000,0.000,0.000|0.000]
[T:3|0.000,0.000,0.000,0.000|0.000]
[T:4|0.000,0.000,0.000,0.000|0.000]
[T:5|0.000,0.000,0.000,0.000|0.000]
[T:6|0.000,0.000,0.000,0.000|0.000]
[T:7|0.000,0.000,0.000,0.000|0.000]
[T:8|0.000,0.000,0.000,0.000|0.000]
[T:9|0.000,0.000,0.000,0.000|0.000]
[T:10|0.000,0.000,0.000,0.000|0.000]
[HOME:-2.000,-2.000,-2.000,0.000:7]
[TLO:0.000,0.000,0.000,0.000]
[PRB:0.000,0.000,0.000,0.000:0]
<Idle|MPos:-6.000,-176.000,-57.005,0.000|Bf:100,1023|FS:0,0,0|WCO:-645.000,-802.000,-145.500,-10.000>
<Idle|MPos:-6.000,-176.000,-57.005,0.000|Bf:100,1023|FS:0,0,0|Ov:100,100,100>
<Idle|MPos:-6.000,-176.000,-57.005,0.000|Bf:100,1023|FS:0,0,0>
<Idle|MPos:-6.000,-176.000,-57.005,0.000|Bf:100,1023|FS:0,0,0>
<Idle|MPos:-6.000,-176.000,-57.005,0.000|Bf:100,1023|FS:0,0,0>
;;; here is is actualy executed, about 0.5 seconds after the click
M101D4E5P6Q1
G4P1
M101D4E5P6Q0
[MSG:MODBUS TX: 04 05 00 05 FF 00]
[MSG:MODBUS RX: 04 05 00 05 FF 00 9C 6E]
[MSG:MODBUS RESPONSE: OK]
<Idle|MPos:-6.000,-176.000,-57.005,0.000|Bf:100,1023|FS:0,0,0>
<Idle|MPos:-6.000,-176.000,-57.005,0.000|Bf:100,1010|FS:0,0,0>
```

Another thing I have experienced is lack of motion smoothnes when I enabled the the verbose logging. According to the status messages from grblhal, the buffer is not filled up properly, when the verbose is enabled. You can see this in the video here https://www.youtube.com/watch?v=M64FJNzDxcA&ab_channel=RichardToth

Version data:
ioSender 2.0.44
Win 11 Pro on Core i5 with 16 GB RAM
connected to grblhal on teensy 4.1 via eth/telnet

terjeio, 2024-09-13:
> I have executed the actual commands as macro:

If you send from the MDI is it still delayed?

The $#/$G output seems strange, I wonder what is causing that. Is it only output when using ioSender?

> Another thing I have experienced is lack of motion smoothnes when I enabled the the verbose logging. 

Do you have _Agressive buffering_ enabled in the app settings? If not does it help to enable it?

risototh, 2024-09-13:
> If you send from the MDI is it still delayed?
This seems to be OK, but I can enter only one command, but the command is executed immediately.

> The $#/$G output seems strange, I wonder what is causing that. Is it only output when using ioSender?
Yes, it's only in ioSender. And it does not occur when the command is entered from MDI, so it seems to be related to the macro.

> > Another thing I have experienced is lack of motion smoothnes when I enabled the the verbose logging.
> Do you have _Agressive buffering_ enabled in the app settings? If not does it help to enable it?
No, it was not enabled, and is seems to solve this issue. Thanks!

closed 2024-10-26 by terjeio