terjeio/ioSender #408: Block Delete Character Invalid when Loading Gcode File

engigeer, 2024-10-10:
I am trying to use the block delete functionality recently added to grblhal. If I understand correctly, enabling this mode via my configured input pin, should then skip execution of all lines of gcode that begin with the block delete character (i.e., forward slash ```/```).

I have verified this function of block delete when sending individual lines of code via the console, but it seems that the block delete character may be failing some validation performed when loading a gcode file.

terjeio, 2024-10-15:
Try the latest [edge builds](https://www.io-engineering.com/downloads) (p12).

engigeer, 2024-10-15:
Files seem to be loading correctly in the p12 edge build. Much appreciated!

closed 2024-11-11 by engigeer