terjeio/ioSender #414: Controller START button ignored

calabr, 2024-11-11:
IoSender 2.0.45.P12 and all 2.0.45
; 1.1ha.20201101
; [OPT:VPR+,15,128]

First press of hardware START button on GRBL often ignored by IoSender.
It's normally decoding button pressed, light-up red "S" on DRO, but doing nothing.
Usually second press of the same button starting current G-code streaming.
  This condition often usually registered after opening program and setting working Zero or probbibg to set TLO offset.
Console output for 2 consecutive button press:
```
<Idle|MPos:-245.828,-98.765,-68.918|Bf:15,128|FS:0,0>
<Idle|MPos:-245.828,-98.765,-68.918|Bf:15,128|FS:0,0>
<Idle|MPos:-245.828,-98.765,-68.918|Bf:15,128|FS:0,0|Pn:S>
<Idle|MPos:-245.828,-98.765,-68.918|Bf:15,128|FS:0,0|Pn:S|WCO:-248.205,-102.765,-68.918>
<Idle|MPos:-245.828,-98.765,-68.918|Bf:15,128|FS:0,0|Ov:100,100,100>
<Idle|MPos:-245.828,-98.765,-68.918|Bf:15,128|FS:0,0>
<Idle|MPos:-245.828,-98.765,-68.918|Bf:15,128|FS:0,0>
<Idle|MPos:-245.828,-98.765,-68.918|Bf:15,128|FS:0,0>
<Idle|MPos:-245.828,-98.765,-68.918|Bf:15,128|FS:0,0>
<Idle|MPos:-245.828,-98.765,-68.918|Bf:15,128|FS:0,0>
<Idle|MPos:-245.828,-98.765,-68.918|Bf:15,128|FS:0,0>
<Idle|MPos:-245.828,-98.765,-68.918|Bf:15,128|FS:0,0>
<Idle|MPos:-245.828,-98.765,-68.918|Bf:15,128|FS:0,0>
...
<Idle|MPos:-245.828,-98.765,-68.918|Bf:15,128|FS:0,0|Ov:100,100,100>
<Idle|MPos:-245.828,-98.765,-68.918|Bf:15,128|FS:0,0>
<Idle|MPos:-245.828,-98.765,-68.918|Bf:15,128|FS:0,0>
<Idle|MPos:-245.828,-98.765,-68.918|Bf:15,128|FS:0,0>
<Idle|MPos:-245.828,-98.765,-68.918|Bf:15,128|FS:0,0|Pn:S>
<Idle|MPos:-245.828,-98.765,-68.918|Bf:15,128|FS:0,0|Pn:S>
<Run|MPos:-245.828,-98.765,-66.480|Bf:0,19|FS:77,0>
<Run|MPos:-245.828,-98.765,-67.963|Bf:0,17|FS:200,0>
<Run|MPos:-245.828,-98.765,-68.633|Bf:0,17|FS:200,0>
```
It's also incorrect state indication if START button pressed during GRPL Alarm state - 
IoSender indicating Alarm, but after presing START button, GRBL returning alarm
`error:9 - G-code lock: G-code commands are locked out during alarm or jog state.`
but IoSender blocking CircleStart and FeedHold, enabling Pause button and still indicating red Alarm machine state sign. 
returning from this state require to push Pause than Stop button.

terjeio, 2024-11-11:
> First press of hardware START button on GRBL often ignored by IoSender.

Is the signal even reaching the sender? Legacy Grbl and older grblHAL versions did not send the status if the hardware button was not pressed when a real-time request was prepared by the controller. A workaround for this is to press the button long enough for it to be reported.

Also, there is an [interlock mechanism](https://github.com/terjeio/ioSender/blob/614115d80c1f6c792112371e94f73741558cc407/CNC%20Core/CNC%20Core/GrblViewModel.cs#L987-L992) in ioSender that suppresses the first cycle start keypress, this is to block accidental motion that was reported by a user (or users). It may be this that is interfering as well, and it is tricky to solve in a safe way. I do not remember the exacrt reason now, but part of it could be due to many not wiring hardware buttons resulting in the `S` (and `H`) state beeing always reported.

> returning from this state require to push Pause than Stop button.

Hmm, I guess harware cycle start should be should be ignored during an alarm?

calabr, 2024-11-23:
> Is the signal even reaching the sender? 
 Yes, sender definitely see this signal - I see red light S indicator in IoSender DRO panel for both first and second button press.

> Hmm, I guess harware cycle start should be should be ignored during an alarm?
  for legacy GRBL button press reported in alarm state.

terjeio, 2024-11-30:
> I think IoSender should follow controller state and adjust UI accordingly, instead of holding it's own state machine in parallel.

If only it was that simple...