terjeio/ioSender #418: Heatmap probbing crashes with exception

calabr, 2024-11-23:
IoSender 2.0.45.P12 and all 2.0.45
; 1.1ha.20201101
; [OPT:VPR+,15,128]

Heatmap probbing with option "set Z=0 at X0Y0; set coord system G54" often cause exception on the last step.
```
---------------------------
DispatcherException
---------------------------
Unhandled exception occured: Index was out of range. Must be non-negative and less than the size of the collection.

Parameter name: index
---------------------------
```
terminal output show the last probbing command completed, but offset not implemented

```Probing point 24 of 25...
G38.2F250Z-10
<Run|MPos:-273.828,-131.523,-88.665|Bf:14,91|FS:35,0>
<Run|MPos:-273.828,-131.523,-88.780|Bf:14,91|FS:35,0>
<Run|MPos:-273.828,-131.523,-88.905|Bf:14,91|FS:0,0|Pn:P>
[PRB:-273.828,-131.523,-88.895:1]
PM:ok
G0Z0.5
PM:ok
G38.2F35Z-2
PM:ok
G53G0Z-86.868
<Run|MPos:-273.828,-131.100,-86.868|Bf:14,95|FS:2142,0>
<Run|MPos:-273.828,-126.523,-86.935|Bf:14,95|FS:250,0>
<Run|MPos:-273.828,-126.523,-87.773|Bf:14,95|FS:250,0>
<Run|MPos:-273.828,-126.523,-88.605|Bf:14,95|FS:250,0>
[PRB:-273.828,-126.523,-88.898:1]
PM:ok
G0Y5.000
PM:ok
Probing point 25 of 25...
G38.2F250Z-10
<Run|MPos:-273.828,-126.523,-88.618|Bf:15,91|FS:0,0>
<Run|MPos:-273.828,-126.523,-88.733|Bf:14,91|FS:35,0>
<Run|MPos:-273.828,-126.523,-88.858|Bf:14,91|FS:35,0>
[PRB:-273.828,-126.523,-88.895:1]
PM:ok
G0Z0.5
PM:ok
G38.2F35Z-2
PM:ok
G53G0Z-86.868
<Run|MPos:-273.828,-126.523,-87.490|Bf:13,95|FS:333,0>
<Run|MPos:-273.828,-122.293,-86.868|Bf:14,95|FS:354,0>
<Run|MPos:-273.828,-121.523,-87.423|Bf:14,95|FS:250,0>
<Run|MPos:-273.828,-121.523,-88.255|Bf:14,95|FS:250,0>
<Run|MPos:-273.828,-121.523,-89.083|Bf:14,95|FS:0,0|Pn:P>
[PRB:-273.828,-121.523,-88.923:1]
PM:ok
<Run|MPos:-273.828,-121.523,-88.688|Bf:14,114|FS:35,0>
<Run|MPos:-273.828,-121.523,-88.805|Bf:14,114|FS:35,0|WCO:-293.828,-141.523,-86.868>
<Run|MPos:-273.828,-121.523,-88.923|Bf:14,114|FS:0,0|Pn:P>
[PRB:-273.828,-121.523,-88.918:1]
<Idle|MPos:-273.828,-121.523,-86.868|Bf:15,128|FS:0,0>
<Idle|MPos:-273.828,-121.523,-86.868|Bf:15,128|FS:0,0|Ov:100,100,100>
Probing...
G0G90X0.000Y0.000
PM:ok
Probing @X0Y0...
G91F250
PM:ok
G38.2F250Z-10
PM:ok
G0Z0.5
PM:ok
G38.2F35Z-2
PM:ok
<Run|MPos:-293.105,-140.800,-86.868|Bf:14,109|FS:566,0>
<Run|MPos:-293.828,-141.523,-87.420|Bf:14,109|FS:250,0>
<Run|MPos:-293.828,-141.523,-88.255|Bf:14,109|FS:250,0>
[PRB:-293.828,-141.523,-88.915:1]
<Idle|MPos:-293.828,-141.523,-89.118|Bf:15,116|FS:0,0|Pn:P>
<Run|MPos:-293.828,-141.523,-88.688|Bf:14,128|FS:35,0>
<Run|MPos:-293.828,-141.523,-88.805|Bf:14,128|FS:35,0>
<Run|MPos:-293.828,-141.523,-88.923|Bf:14,128|FS:0,0|Pn:P>
[PRB:-293.828,-141.523,-88.910:1]
<Idle|MPos:-293.828,-141.523,-88.938|Bf:15,128|FS:0,0|Pn:P>
<Idle|MPos:-293.828,-141.523,-88.938|Bf:15,128|FS:0,0|Pn:P|Ov:100,100,100>
<Idle|MPos:-293.828,-141.523,-88.938|Bf:15,128|FS:0,0|Pn:P|WCO:-293.828,-141.523,-86.868>```

terjeio, 2024-12-11:
> ; 1.1ha.20201101

This is not the original Grbl, the last version dates for that are 20190830 and 20180813.Mega. Do you have a link to the source code for the one you use?

I will fix the exception, it is due to a probing failure and a missing check for probe position aquired for the last point. The fix will ensure it ends with a failure feedback instead.
Why it fails is another matter, I have not been able to track that down just yet.

calabr, 2024-12-13:
> Do you have a link to the source code for the one you use?

 Yes - It's 1.1f modified according to your great recommendation for proper jogging.
  [https://github.com/calabr/grbl/tree/door-setup](https://github.com/calabr/grbl/tree/door-setup)