terjeio/ioSender #421: Saving settings with ethernet error

phil-barrett, 2024-12-02:
Edge version 2.0.45p12 and release version 2.0.44

I have a grblHAL board with optional ethernet adaptor (PicoCNC).  When using firmware with W5500 Ethernet selected but the physical hardware not present, I see odd behavior in ioSender.  Specifically, if I change a Grbl setting in the Settings: Grbl tab and click on Save settings, ioSender gives an error "Failed to start ethernet stack" in the lower left corner. Then, when I click on the Grbl tab, it gives me an error "Changes not saved, Save now?".  I click yes but it does not go back to the Grbl Tab.  If I click on the Grbl Tab, it stays in the settings tab.  I can click on any other tab and it shows that tab.  Then, I can click on the Grbl tab and it shows it.  Going back to the Settings: Grbl tab shows that the changes were saved.

With the wiznet ethernet adaptor installed, there is no "Failed to start..." error and ioSender (44 and 45p12) behaves as expected.

terjeio, 2024-12-11:
Currently I attempt to init the adaptor/network stack on settings changes, I will change this to only on controller startup in the next commit.
This will fix the issue.

terjeio, 2024-12-12:
[Fix committed](https://github.com/grblHAL/RP2040/commit/cd4fbfbd7469ab481300588ff34b77cb826263ef).