terjeio/ioSender #424: v2.0.45 Settings backup not writing a settings.txt file

axg20202, 2024-12-13:
Not sure if it's an issue specific to my machine, but the previous release saved the settings backup with no issue. Latest release gives the usual message that a settings.txt file was saved, but it's not in the folder.

phil-barrett, 2024-12-13:
Did you use the Backup button or the Save button? With Backup, I get a settings.txt file in the directory.  Another possibility, does the directory require Administrator privilege?

axg20202, 2024-12-13:
Definitely clicking backup. Admin privileges for the folder could well be it. Will double check.

phil-barrett, 2024-12-13:
I've never quite understood how the whole admin thing works in Windows when you try to write to something that you don't have privilege to. I think a lot of apps fail silently. Is there a failure notification that happens at the application level?  If it does, it would be really nice for ioSender to notify the user - could make it easy to solve related problems.

axg20202, 2024-12-14:
It is a bit strange because I don't recall changing the program folder permissions for the previous release, and it's in the same place.

boboxx, 2025-01-27:
I'm also having issues with my settings. If I add a new probe profile, set my settings and close ioSender my profile is gone. Is there a was to same probing profiles in version 2.45? I also tried 2.44 and have the same issue.

terjeio, 2025-02-13:
I am not able to replicate this.
Basic app settings and macros are stored in _App.config_ and probing profiles in _ProbingProfiles.xml_ in the application folder by default. The App settings file to be used can be placed elsewhere by adding a [command line parameter](https://github.com/terjeio/ioSender/wiki/Setup-and-configuration#optional-command-line-parameters), I probably should add another for which folder to use for _settings.txt_,  _ProbingProfiles.xml_ and the other xml files used for configuration parameters.

> If it does, it would be really nice for ioSender to notify the user - could make it easy to solve related problems.

Agree, I'll add that - currently I have empty catch blocks so saving will fail silently.

boboxx, 2025-02-13:
This is likey related to UAC, i'mm pretty sure i still have enabled on my computer and I have moved the application to the Program Files folder. I will report back.

terjeio, 2025-02-14:
[New edge version](https://github.com/terjeio/ioSender#edge-pre-releases) uploaded.