terjeio/ioSender #425: Door intercation  breaks Manual Toolchange at G59.3

mintracer, 2024-12-13:
Hey Terje, 

i hope your Holiday was relaxing! :)

I have added some doorswitches to my enclosure, when i now run a gcode with a toolchange, then everything works until the first toolchange is requested...

a small step by step:

the toolchange is requested : the spindel rises and goes over G59.3 and waits...
then i open the door, the "DOOR:1" appears and i start to change the endmill... 
now the endmill is changed, i closed the door... and now evrything is broken...

i can´t press start -  in the Console comes: "First do a "$TPW" for probing... ", i have a Macro with "$TPW", also not working... i can´t do anything, except "STOP", when i press "STOP" the running Gcode and offsets are useless ... 

So what can i do? 

Today i broke an endmill, thinking that a reprobing and restarting at a specific line... the spindle went down with G0... 

If you have further questions, just ask... 

greeting from Germany :)

Einar

mintracer, 2024-12-13:
here a small video of it...

https://foto-shoot.de/nextcloud/index.php/s/o2SERe3QZjLTdFK

sorry for my bad english and my voice... i´m speaking without voicebox :)

Thx for watching

terjeio, 2024-12-13:
Hmm, I guess the door signal should be suppressed while changing tools. I did not think of that earlier.
It is a bit like the option to suppress it when in IDLE state, I enable that so I can position my CO2 laser with the lid (door) open.

mintracer, 2024-12-13:
maybe just add the "TOOL" state to the section where the "IDLE" state is ignored..

terjeio, 2024-12-14:
> maybe just add the "TOOL" state to the section where the "IDLE" state is ignored..

Yes, and also add a check for door open after the tool change has been completed. So best to close the door before ending it with Cycle start.

> is this ioSender related or controller based?

Controller. I'll commit a fix soon.

mintracer, 2024-12-14:
thank you very much!

terjeio, 2024-12-14:
Fix committed and Web Builder updated. Please report back if it works as intended.

mintracer, 2024-12-14:
Hey Terje...

you are really awesome... 

in the webbuilder i get this message...

Command failed: pio run -etest -cjob.ini
src/grbl/ngc_flowctrl.c: In function 'ngc_flowctrl':
src/grbl/ngc_flowctrl.c:694:23: warning: 'sub' may be used uninitialized in this function [-Wmaybe-uninitialized]
if(sub == NULL)
^
.pio/libdeps/test/grblhal-rgb-plugin/status_light.c:207:21: error: static declaration of 'bit_count' follows non-static declaration
static uint_fast8_t bit_count (uint32_t n)
^
In file included from src/grbl/config.h:112:0,
from src/grbl/grbl.h:37,
from src/grbl/hal.h:31,
from src/grbl/driver_opts.h:30,
from src/driver.h:48,
from .pio/libdeps/test/grblhal-rgb-plugin/status_light.c:36:
src/grbl/nuts_bolts.h:233:14: note: previous declaration of 'bit_count' was here
uint_fast8_t bit_count (uint32_t bits);
^
.pio/libdeps/test/grblhal-rgb-plugin/status_light.c: In function 'rgb_state_changed':
.pio/libdeps/test/grblhal-rgb-plugin/status_light.c:427:76: warning: implicit declaration of function 'ffsl' [-Wimplicit-function-declaration]
if(bit_count(hard_limits.mask) == 1 && (axis = ffsl(hard_limits.mask)) != -1)
^
*** [.pio/build/test/libf0e/grblhal-rgb-plugin/status_light.c.o] Error 1
========================= [FAILED] Took 30.63 seconds =========================

its related to the RGB plugin...

Thx for looking!

And please take it easy!

Einar... :)

terjeio, 2024-12-14:
Try again now

mintracer, 2024-12-14:
Firmware was created successfully!

thank you very much!!

what a great support!!

mintracer, 2024-12-15:
so firmware flashed and initial test done (rerun the program from yesterday...)

the Door system is ignored by the TOOL state so that works now!

Only the RGB Pins are different after rebuilding the Firmware.
the assignment has changed so the Colors are messed up... :) 

the Pins are assign as follows. 
PIN0: 0
PIN1: R
PIN2: G
PIN3: B

but thats another thing... the main important thing was solved in just 24h... amazing!!!

Thank you!

Cheers 

Einar

closed 2025-02-07 by terjeio