terjeio/ioSender #427: Object reference not set to an instance of an object

stevempotter, 2024-12-18:
"Object reference not set to an instance of an object" I have no idea what that means. 
[Drilling_holes.zip](https://github.com/user-attachments/files/18188363/Drilling_holes.zip)
I get the above Unhandled Exception error just after loading in a .nc file that was generated by Fusion using their latest grbl post processor. IOsender promptly quits.
I am using v. 2.0.45XL.
Any advice about what is the problem and how to fix it?
I will try to attach the gcode file, which is just to drill 3 holes.

terjeio, 2024-12-25:
"just after loading" means that you do not have any interaction at all with the sender after opening the file?
Asking since when I try I can load and run it with no issues.

stevempotter, 2024-12-25:
Yes, the error dialog pops up immediately, and I can't do anything but dismiss it, then IOsender exits.

terjeio, 2024-12-26:
Does it still happen if you remove the G28 line (line 6)?

stevempotter, 2024-12-27:
Sorry, now I can't reproduce the error either; with or without that line, it works. If it is not homed yet, a warning message pops up that it has G28 commands and is not homed, but it allows me to dismiss that without crashing. I will let you know if I get the problem again.

closed 2025-02-07 by terjeio