terjeio/ioSender #434: Error 4 on Center Finder

MartinBaadsgaard, 2025-01-14:
Using Center Finder in 2.0.45, I get Error: 4 after first move of the 4th probe, 1st pass.
By suggesting, tried disabling soft limits.
No error in 5 runs with soft limits off, 3/5 with limits on.

[image]

terjeio, 2025-01-28:
Fix available for you to test in the latest [edge version](https://github.com/terjeio/ioSender#edge-pre-releases).

MartinBaadsgaard, 2025-01-28:
Thank you!
Tried ioSender Edge 2.0.46p1, unfortunately I got the same error again at different points in the probing process (2nd pass 2nd probe, 2st pass 2nd probe)
Soft limits on:
Repro rate: 2/3
Soft limits off:
Repro rate: 1/3

I have also found an error in my initial reporting. (Very sorry about that!) It was ALARM 4, not error 4.
I hope this is not a massive difference in how you interpreted the issue.

Let me know if I can provide anything else. Videos, pictures, system specs.

terjeio, 2025-01-29:
Does it help increasing the _Latch distance_?
The probe motion is decelerated to a halt on contact, this takes time. There is a compiler option for reducing it if you need a small latch distance, if using the Web Builder to generate the firmware you can find it in the _Advanced features_ tab, Minimize probe overshoot_ checkbox. This may also be worth a try.

terjeio, 2025-01-30:
> As a sidenote, issue https://github.com/terjeio/ioSender/issues/435 occurred consistently through testing, with probe stopping one move before finishing.

With 2.0.46p1? I was able to reproduce the behaviour with 2.0,45 but I have not seen it again after more than one hundred runs with 2.0.46p1.

MartinBaadsgaard, 2025-01-30:
Sorry, I did not realise it was that bug which had been addressed, the release comment only mentioned this one.
Let me test on 2.0.46p1 when I'm back at the machine.