terjeio/ioSender #437: Probe ends at X= +/- 0.005 after probing Center

MartinBaadsgaard, 2025-02-02:
Using Center Finder always ends with the probe at X=0.005 both in 2.0.45 and 2.0.46p1.
This leaves it ambiguous whether the probe is at center but WCS is off, or WCS is on center but probe is off.

Tried different latch distances (0.5-2) and XY clearance (5-15) and rapids feeds (1000-3000)

(Of note here; this is not an immediate issue for me personally as this is below my expected tolerances. If preferred I can close the issue again).

terjeio, 2025-02-02:
What is you steps/mm resolution?

MartinBaadsgaard, 2025-02-02:
Motor step angle is 1.8° on 1610 ballscrews, so 200 steps/mm

MartinBaadsgaard, 2025-02-02:
Yep, no worries. It was consistently off so thought there was potentially a missed step or something and thought I'd raise it.
Closing ticket.

closed 2025-02-02 by MartinBaadsgaard