terjeio/ioSender #440: Error 20 with Turning gcode

JeffHank1, 2025-02-28:
I am trying to use the Turning feature in iOSender. The gcode generates quickly for a simple diameter reduction operation but stops almost immediately with an Error 20. The line of code is: G18 G8 G21
I'm using ioSender 2.0.45. I've also tried it with ioSender 2.0.44. Control card is a MKS TinyBee V1.0

The entire code for the operation is.
G18 G8 G21
M3S100 G4P1
G0 X7.750
G0 Z0.500
G97
(Pass: 1, DOC: 5.673 0.077)
G1 X5.673 F200.000
G1 Z-10.000 F200.000
G0 X7.673
G0 Z0.500
(Pass: 2, DOC: 5.596 0.077)
G1 X5.596 F200.000
G1 Z-10.000 F200.000
G0 X7.596
G0 Z0.500
(Pass: 3, DOC: 5.520 0.077)
G1 X5.520 F200.000
G1 Z-10.000 F200.000
G0 X7.520
G0 Z0.500
(Pass: 4, DOC: 5.500 0.020)
G1 X5.500 F100.000
G1 Z-10.000 F100.000
G0 X7.500
G0 Z0.500

terjeio, 2025-02-28:
Your controller does not support G8?
Or if grblHAL is _Lathe mode_ enabled by setting $32?

[image]

JeffHank1, 2025-02-28:
$32. That was it. Thank you. seems to run perfectly now. Thanks again for a great application

JeffHank1, 2025-02-28:
The program seems to want to stay in Radius Mode. Changing from Disabled to Diameter mode under Settings:App then Saving doesn't seem to change anything. The Grbl DRO stays at Radius Mode and movement tests stay at Radius mode. Is there another setting required to change this?

terjeio, 2025-02-28:
Do not set lathe mode in _Settings: App_, that is for non-grblHAL controllers. G7/G8 is for controlling Diameter/Radius mode.

JeffHank1, 2025-02-28:
Yes. I set Lathe mode in Grbl Settings. 
Different problem. I couldn't find a setting for Diameter Mode. iOSender seems to stay in Radius mode

JeffHank1, 2025-02-28:
Does a code need to be specified in the Turning code to implement anti backlash or does GrblHal just do it?  
I have noticed my turning diameters and length are off by exactly the amount of backlash I entered under Settings:Grbl, that is X 0.28mm too big and Z 0.78mm too short

terjeio, 2025-03-01:
grblHAL should do it, sound like backlash base position is on the wrong side of motion. Do you home the axes?

JeffHank1, 2025-03-01:
I don't home the axes so I guess I'll have to try that, or move the cutter away from the 0 points before cutting. Thanks.

terjeio, 2025-03-02:
Backlash compensation is enabled on controller startup or on homing completed.
If it works correctly or not after startup (before homed) depends on which end of the backlash the machine is.
FYI the homing direction setting ($23) is used to set the initial direction.

JeffHank1, 2025-03-02:
Thanks for the comments. I'll play with it more today. If all else fails I'll install homing end stop switches.

JeffHank1, 2025-03-02:
Without a Homing cycle, accuracy with Backlash seems to work best by running both X and Z in a negative direction to reach the 0,0 starting point, ie moving both axes in a Positive direction first then running back in a Negative direction to the the starting point on the Lathe work piece. For example, getting to the start point for Turning an accurate radius before threading. This would be the same as a Homing motion I suppose.

For threading since iOSender seems to try to do this under power, I switch to Grblgru and thread under Stepper control of the C/A axis. I can input RPM and Sync signals in a different program (CNCL) using a Mega 2560 board and 2 Hall effect sensors. 

For fun I gave it a try with iOSender anyway. Perhaps I missed some variables. My Spindle is Disabled under Settings:Grbl since I don't think there are RPM and Sync input pins for my MKS TinyBee V1.0. 

P   Depth Cut    Area
01 0.100  0.100 0.006
02 0.200  0.100 0.017
03 0.300  0.100 0.029
04 0.400  0.100 0.04
05 0.500  0.100 0.052
06 0.600  0.100 0.064
07 0.700  0.100 0.075
08 0.800  0.100 0.087
09 0.900  0.100 0.098
10 1.000  0.100 0.11
11 1.013  0.013 0.015

1	ok	(M 10 x 1.5)
2	ok	G18G8G21
3	ok	M3S100G4P1
4	ok	G0X5.175
5	ok	G0Z0.500
6	error:20	G76P1.500Z-10.000I-0.250J0.100K1.013
7		M30

I also tried the Facing function. If I enter a starting Z of 0 and a target Z of 1 I get a red box around them both when the Calculate button is clicked. It doesn't seem to matter if I enter a constant diameter or a decreased diameter setting. I leave the other settings to default values.