terjeio/ioSender #442: Can someone help on creating an embroidery specific version of IOsender

Ams-Laser, 2025-03-07:
here is a video of my current machine working, i would love to have a special sender for it with some extra features
https://youtube.com/shorts/ccW4PQBU1mk?si=Whr4XTzKCeQr84WK
thanks
Ed

terjeio, 2025-03-13:
What are these extra features?

Ams-Laser, 2025-03-14:
> What are these extra features?

tread break detection, composite stepping syncronized to needle camshaft, use of pyembroidery library to accept many more than just DST files, post procesing to specific machine commands, beeing able to rewind or ffwd code from a pause, 
and some more, i would be willing to fund such a project, so far i have written some glue-ware that does almost all that is needed, and converts the embroidery file to a preview and assigns neddles to colors, post proces the dst file to custom gcode and uploads and run from SD on either grbl-hal or fluidnc running on a esp32, it would be better off to just modify directly the firmware to process inputs from neddle, position encoder, and tread pulls, to make a embroidery machine specific firmware, as still tread cutting and tread wiper support is incomplete, you can reach me at cnclasermx at gmail you know the rest, and we can discuss a work like such
thanks
Ed

terjeio, 2025-03-17:
> so far i have written some glue-ware that does almost all that is needed

Is that available anywhere?

I do not believe it will be possible to run an advanced embroidery machine from gcode, too many real time signals to handle for that. And is is the reason I opted to create the [grblHAL plugin](https://github.com/grblHAL/Plugin_embroidery/) which bypasses gcode and controls the machine directly via functionality provided by the core and associated driver. My approach translates embroidery files, [like this](https://github.com/grblHAL/Plugin_embroidery/blob/main/brother.c), to a common command format that is then used to control the machine. And FYI I do not want to do development work on the ESP platform, it is way to expensive and limited in I/O capability, RP2040/RP2350 or STM32Fx based MCUs are a lot cheaper to develop for. Another issue is having access to hardware (or a simulator) to test/verify with...

To add more advanced capabilities to the grblHAL plugin either detailed specifications with timing diagrams, or a developer with access to hardware that I can guide, is required. I would prefer the latter.

If FluidNC support is required you will have to bring a developer familiar with that codebase onboard as well.

Ams-Laser, 2025-03-17:
> Is that available anywhere?
>
> I do not believe it will be possible to run an advanced embroidery machine from gcode, too many real time signals to handle for that. And is is the reason I opted to [...]

terjeio, thanks for your response, beliveit or not i now have a brother BAS-415 industrial machine working on Gcode, there ate two ways of handling movement on embroidery machines, i have repaired electronic boards for industrial machines for quite some time, an there is really a much simpler logic than most people imagine, if possible contact me via whatsapp and we can discuss how it can be done, and i would be happy to provide my code for you to base something better at, mabe a custom firmware for embroidery machines, you can reach me via whatsapp  at +523311196406