terjeio/ioSender #443: Request - include the alarm or error description.

LoganFraser, 2025-03-10:
In addition to the error or alarm #, display that error or alarm message in console. this is potently the most common question asked about grblhal/iosender in the PrintNC discord (looking in help is apparently not intuitive enough)

terjeio, 2025-03-13:
I have added description to the alarm code in the console and the text in the status line in the lates [edge version](https://github.com/terjeio/ioSender#edge-pre-releases).

Ther was a "bug" in that the sender attempted to clear some alarms automatically - this has bee fixed so alarm state and related message in the status line will no longer be cleared.

LoganFraser, 2025-03-15:
Nice!  Thanks for the update.

closed 2025-03-15 by LoganFraser