terjeio/ioSender #444: Minor UI requests - Change radio buttons to toggle text buttons and others.

LoganFraser, 2025-03-16:
UI request changes: 

Swap the radio buttons for Spindle Off - CW - CCW to full text buttons with active mode - similar to the Home button.  The radio buttons are very difficult to click on a touch screen, especially because they're vertically aligned and close together. 

Swap the check boxes for Mist and Flood to full text buttons with active mode - as above.

Store previous manual RPM setting change rather than resetting to min RPM after every stop, or have a setting option for it. I could just make a macro to start at 12k or 16k or 20k but it's easier to just type the number in the field - but it disappears again when the spindle is stopped between manual cuts. 

I really appreciate all the efforts you have put into both IOSender and Grblhal.  Thanks!

LoganFraser, 2025-03-18:
An additional request - add support for a TS.macro (toolsetter) so any calls to measure tool length use this if it exists, same as the function of TC.macro. helpful if there's a cover over the toolsetter or it's on an arm or slider to move outside the cutting area.

phil-barrett, 2025-05-06:
> > Store previous manual RPM setting change rather than resetting to min RPM after every stop, or have a setting option for it. I could just make a macro to start at 12k or 16k or 20k but it's easier to just type the number in the field - but it disappears again when the spindle is stopped between manual cuts.
> 
> It does not for me. What is your workflow?

For me, on both Win 10 and 11, the RPM box retains the previously set value and shows 0 when set to Off.  It does not retain the previous setting when restarting ioSender.  

Note that if you use MDI to send M3Sxxx, the RPM box show the previous setting, not the S value.  I have not measured it to see what is actually happening with the spindle but suspect it reflects the Sxxx value and not the RMP box value. And, a special case, if you never fill out the RPM box, it shows 0 after MDI M3S475 as in the screenshot below. When running a GCode program, ioSender greys the spindle section, perhaps it should grey the RPM box between M3 and M5?

[image]

Robin-SeaThrough, 2025-06-05:
Regarding Spindle RPM, could it possible to have two boxes for the Spindle RPM? As in one input box (Requested or Programmed RPM) for manual entry and showing what the Gcode sends. And one Actual RPM, showing RPM feedback from a VFD or similar when available?

engigeer, 2025-09-02:
> New [edge version](https://github.com/terjeio/ioSender#edge-pre-releases) now up with toggle buttons for coolant:
> 
> [image]
> 
>
Could you confirm if [this](https://github.com/ejensen/toggle-switch-control?tab=readme-ov-file) is the library you are using for the toggle buttons? It looks correct and matches the name format for the reference, but for some reason I’m getting a compilation error related to a missing **click** property.

terjeio, 2025-09-03:
@engigeer it is the library I use, and I have wrapped it in a user control that provides the click event:

[ToggleControl.zip](https://github.com/user-attachments/files/22109104/ToggleControl.zip)

engigeer, 2025-09-04:
Thanks! As far as I can tell, the original library does not expose **Click** as a public event. The wrapper ToggleControl attempts to subscribe to `tsw.Click`, but seems to cause a compilation error because the event does not exist. Is it possible you are using a custom version of the library?

terjeio, 2025-09-04:
> Is it possible you are using a custom version of the library?

Seems so, but I cannot remember changing it... This is the changed file:

[ToggleSwitchBase.zip](https://github.com/user-attachments/files/22133799/ToggleSwitchBase.zip)

I am referencing the version [downloaded from git](https://github.com/ejensen/toggle-switch-control).

engigeer, 2025-09-04:
> Seems so, but I cannot remember changing it... This is the changed file:
> 

Ah perfect. Everything is compiling as expected now. Thanks!