terjeio/ioSender #445: ethernet connection dies

Ed-27, 2025-03-20:
If I let the ethernet connection between iosender and flexihal go idle while preparing stock they will stop communicating.  The only way to restart the connection is to restart both Iosender and flexihal.  When I bring them back up Iosender will display a message that the CNC needs to homed. Is there to prevent this disconnect or retain my work offset G55 or G56 .

Thanks Ed.

phil-barrett, 2025-03-20:
I would try to figure out why the connection drops.  Maybe your PC is timing out and sleeping.

Anyway, you can turn off **Homing on startup required** in the **Homing cycle**  setting ($22)

Ed-27, 2025-03-21:
Thanks, it was the PC going to sleep. 
When I look at the homing options in IOsender 22 is labeled "Homing Cycle : enabled (33)" which is confusing.