terjeio/ioSender #450: ioSender stuck on startup

Semphora, 2025-04-21:
Heya,
I've noticed ioSender beeing stuck and blocking the rest of the system while waiting for reply of the grbl board during startup. is this intended? or a bug? 
I'm running ioSender XL 2.0.45 on a Win10 system

Semphora, 2025-04-22:
I've deleted the App.config.
The behaviour still persists.
I do not get a "No response received from controller, exiting" message.
But, after roughly 5 minutes (+/20seconds) the program stops blocking the other windows and acts as I would have expected. Displaying 

[image]

While testing this I also noticed, that I was able to open programs via keyboard shortcut and use them as expected, once I had the window focus on them. However, clicking on ioSender would lock the focus back up on ioSender.

I'm not 100 percent on the board. But I do belive it to be an grbl controler. It came with the CNC and has the "ANNOY TOOLS" logo on the top right. It is however, not USB C, so I'm not sure, which version.

phil-barrett, 2025-04-22:
A couple of things to try:
- Open the Device Manager and look in the ports section. what do you see?  
- Use a terminal app, like PuTTY can try to connect to COM8.  Try different baud settings.  Try typing $ or $$ and see what happens.  

ANNOY is a terrible choice of a product name, BTW.

Semphora, 2025-04-22:
Hey Phil, thanks for your input. This is really just a startup issue. The COM Port works and once the board replies everything does work as intended.

terjeio, 2025-04-22:
Could it be the serial port (USB) driver that causes this?
Which is it and is it up to date?

phil-barrett, 2025-04-23:
My point about using a terminal program is to see what the board outputs.  It should have a boot up banner and config info.  All Grbl variants do that.  It's the first step to understanding what you have. If it doesn't output anything, that would be the heart of the problem, no?

Semphora, 2025-04-23:
> My point about using a terminal program is to see what the board outputs. It should have a boot up banner and config info. All Grbl variants do that. It's the first step to understanding what you have. If it doesn't output anything, that would be the heart of the problem, no?

I mean, not really, I guess? If I understand terjeio correctly, the programm should not freeze up during startup in any case. So whether the Controler board replies should be immaterial.

Anyway the Putty Output is:

[HLP:$$ $# $G $I $N $x=val $Nx=line $J=line $SLP $C $X $H ~ ! ? ctrl-x]
ok
$0=10
$1=25
$2=0
$3=6
$4=0
$5=0
$6=0
$10=1
$11=1.000
$12=0.002
$13=0
$20=1
$21=1
$22=1
$23=3
$24=25.000
$25=500.000
$26=250
$27=2.000
$30=1000
$31=0
$32=0
$100=800.000
$101=800.000
$102=800.000
$110=2000.000
$111=2000.000
$112=100.000
$120=20.000
$121=20.000
$122=20.000
$130=500.000
$131=500.000
$132=200.000
ok

Semphora, 2025-04-23:
> Could it be the serial port (USB) driver that causes this? Which is it and is it up to date?

IT's the USB-SERIAL CH340 driver from wch.cn.
Version: 3.9.2024.9 from 6/09/2024

Other Gcode Sender programms don't experience the freeze, so I'd be hesitant to assume it's a driver issue.

Also, when using it in the correct order of operations,the setup does work. It's really just the freezing when not connected to a machine, that seems to be not working as intended.

terjeio, 2025-06-02:
Since I do not have a board to test with and neither able to find the firmware or the source code for the controller I am closing this.