terjeio/ioSender #458: Lathe mode

LaurentLM, 2025-05-12:
Hi Terje,

Thanks for this wonderful piece of software. It's really a great tool.
I spotted a few of "issues" (or things I cant make sense of). Tested on ioSender Edge 2.0.46p8, but I observed the same behaviour on 2.0.45

1. I selected "diameter mode" in the App settings (using a CNC lathe), but this does not seem to change anything. 
The text above the DRO still reads Radius and X movements are not divided by 2.

2. When using the lathe wizards, it would be very useful to have the option to specify the spindle direction. At the moment, the output spins the spindle clockwise, which may be an issue if the lathe is setup as a conventional manual lathe.
I tried to edit the text output from the command generation (changing `M3` into `M4`), and pressed Tab (or enter). This results into an error message and the app crashed:
`"DispatcherException
Unhandled expection occured: The method or operation is not implemented."`

3. Not specific to lathe mode, but I am struggling to tweak the feedrate on continuous jogging. I run the app with the command line flag `-jogmode 1`. The wiki says the slow, normal and fast feedrates are then inherited from the keypad plugin. The feedrates I get seem fixed (normal 200, fast 500mm/min), and do not match GRBL parameters $50, $51, $52.
I find the keyboard continuous jogging super useful and it could be useful to have the option to enable it in the App settings. 

4. Last one, and probably a bit a detail, but would it be possible to validate a number entry in the DRO with Tab as well as with Enter?

terjeio, 2025-05-15:
1. Lathe settings in App settings are defaults for the profile dialog in the wizards. Set $32=2 in the controller to enable the lathe mode.
2. I have added option to select spindle direction in the latest [edge builds](https://github.com/terjeio/ioSender#edge-pre-releases). The code previews are made read-only to prevent the crash - if you want to edit the generated code you have to save and reload it.
3. I'll have to check this out.
4. Maybe - I'll put it on my todo list.

LaurentLM, 2025-05-16:
1. My microcontroller has $32=2. For some reason I imagined that setting the app in diameter mode would divide/multiply by 2 the X axis of the DRO. I have now found that this can be done by sending G7 / G8. It makes a lot more sense do to it that way. No action needed, I just misunderstood what this setting was for.
2. Thanks, I've just downloaded it and it's exactly what was needed.
3. I have done a bit more digging and found that my jog values come from the config file, specifically here:
```
<Jog>
    <Mode>Keypad</Mode>
    <KeyboardEnable>false</KeyboardEnable>
    <LinkStepJogToUI>true</LinkStepJogToUI>
    <FastFeedrate>500</FastFeedrate>
    <SlowFeedrate>200</SlowFeedrate>
    <StepFeedrate>100</StepFeedrate>
    <FastDistance>500</FastDistance>
    <SlowDistance>500</SlowDistance>
    <StepDistance>0.05</StepDistance>
  </Jog>
  <JogUiMetric>
    <Feedrate0>5</Feedrate0>
    <Feedrate1>100</Feedrate1>
    <Feedrate2>500</Feedrate2>
    <Feedrate3>1000</Feedrate3>
    <Distance0>0.01</Distance0>
    <Distance1>0.1</Distance1>
    <Distance2>1</Distance2>
    <Distance3>10</Distance3>
  </JogUiMetric>
```
I'll keep exploring this and report here.
4. Thanks :-)

LaurentLM, 2025-06-20:
Hi,
I just noticed that the CW/CCW option is not available in the treading wizard, any chance you could add that in a future release?
(I tried to do it myself, but setting an environment for C# proved more difficult than I thought).

terjeio, 2025-06-21:
I can do that - but will have to verify it with my lathe. It may take some time since I am am quite busy - several private projects will get priority for a while.