terjeio/ioSender #460: 2.0.46p9 exception on startup

calabr, 2025-05-22:
Hello @terjeio 
2.0.46.p9 show unhandeled exception after startup and trying to read controller parameters

[image]
it's see the current position reported correctly, alarm state ( hot homed) but don't allow to see console etc. application closed after closing exception window.
the same behaviour with known good config from previous version or starting without config file and creating it on startup.
no such problem seen on 2.0.46p8
 mahine is GRBL 1.1h with slight modification for jogging.
the screen resolution is also may influence - it's 1024x768 screen, by default app starting in window mode.
 UPD: with grblHAL it's running normally

```
Grbl 1.1ha ['$' for help]
[MSG:'$H'|'$X' to unlock]
Alarm:-1 - 
[VER:1.1ha.20201101:]
[OPT:VPR+,15,128]
[G54:-237.926,-123.049,-41.993]
[G55:-254.970,-71.417,-71.755]
[G56:-140.164,-148.264,-42.895]
[G57:0.000,0.000,0.000]
[G58:-100.000,-100.000,-60.000]
[G59:-144.000,-91.000,-82.800]
[G28:0.000,0.000,0.000]
[G30:0.000,0.000,0.000]
[G92:0.000,0.000,0.000]
[TLO:0.000]
[PRB:0.000,0.000,0.000:0]
[VER:1.1ha.20201101:]
[OPT:VPR+,15,128]
[MSG:Caution: Unlocked]
[VER:1.1ha.20201101:]
[OPT:VPR+,15,128]
```
 
<details><summary>Details</summary>
<p>

```
;
; 2025-05-22T23:40
;
; 
; 1.1ha.20201101
; [OPT:VPR+,15,128]
;
$N0=
$N1=
; 0 - Step pulse time
$0=5
; 1 - Step idle delay
$1=255
; 2 - Step pulse invert
$2=0
; 3 - Step direction invert
$3=7
; 4 - Invert step enable pin
$4=0
; 5 - Invert limit pins
$5=0
; 6 - Invert probe pin
$6=0
; 10 - Status report options
$10=3
; 11 - Junction deviation
$11=0.010
; 12 - Arc tolerance
$12=0.002
; 13 - Report in inches
$13=0
; 20 - Soft limits enable
$20=1
; 21 - Hard limits enable
$21=0
; 22 - Homing enable
$22=1
; 23 - Homing direction invert
$23=1
; 24 - Homing locate feed rate
$24=20.000
; 25 - Homing search seek rate
$25=1000.000
; 26 - Homing switch debounce delay
$26=20
; 27 - Homing switch pull-off distance
$27=1.000
; 30 - Maximum spindle speed
$30=12000
; 31 - Minimum spindle speed
$31=0
; 32 - Laser mode
$32=0
; 100 - X-axis travel resolution
$100=400.000
; 101 - Y-axis travel resolution
$101=400.000
; 102 - Z-axis travel resolution
$102=400.000
; 110 - X-axis maximum rate
$110=4000.000
; 111 - Y-axis maximum rate
$111=4000.000
; 112 - Z-axis maximum rate
$112=4000.000
; 120 - X-axis acceleration
$120=400.000
; 121 - Y-axis acceleration
$121=400.000
; 122 - Z-axis acceleration
$122=400.000
; 130 - X-axis maximum travel
$130=348.000
; 131 - Y-axis maximum travel
$131=246.000
; 132 - Z-axis maximum travel
$132=108.000
```
</p>
</details>

terjeio, 2025-06-02:
New [edge pre-release](https://github.com/terjeio/ioSender#edge-pre-releases) with fix now available.

calabr, 2025-06-02:
verified with 2.0.46.p10 - now it's starts normally
Other issue found - show working area boundary button stopped to work on 3D View page, but can be switched from IoSender options page
new override controls are not very robust - I'm using touchpad and it's not comfortable to use slider, but if I'm trying to tap 2 or 3 tomes left or right on slider -it's not always jumping right amount - some taps are missing.  probable it's trying to use double-click.