terjeio/ioSender #467: Probing - Offset travel on successful probing

EmpyreanCNC, 2025-06-04:
Offset clearance on the edge finding probing tab is greyed out, but it isn't on the rotation probing. 

In order to have a more accurate rotation probing routine you want to set this value as high as is possible that can be done on the relevant geometry

But if you run a rotation probe, then switch back to a simple edge find it remembers this offset value

In the edge finding probing the probing cycle runs and on successful probe then travels that offset distance away from the detected edge. As a result if you have a cramped workspace it makes it very easy to crash your spindle.

Video of the behaviour here:

https://www.youtube.com/watch?v=w45Ot4CHcq8

I'm not versed in C# but i think the issue might lie here in the screenshot, Line 190

[image]

Taken from here, https://github.com/terjeio/ioSender/blob/master/CNC%20Controls%20Probing/CNC%20Controls%20Probing/EdgeFinderControl.xaml.cs#L190

I think the - probing.Offset on line 190 should be removed to make the probe tip only travel back to the probing.startposition keeping it consistent with the entry of the probing move

terjeio, 2025-06-05:
It is a bug, the line should be:

`rapidto.Values[axis] = probing.StartPosition.Values[axis] - XYClearance * af[axis];`

If `probing.Offset` is removed the probe will likely crash into the workpiece - depending on the starting position.

I am not yet decided if I should create another edge build or commit a full release...

Robin-SeaThrough, 2025-06-05:
@terjeio is probing.Offset used for anything other than the distance between 2 points along an edge when using Rotation probing?

If it is, I would very much prefer if that could be split into 2 different offsets, to atleast make sure that Rotation has its own dedicated offset.

I've had the same issue as @EmpyreanCNC, managed to crash my probe into a vise when doing an edge finding only along Y, right after probing rotation along X with an 80mm offset 😅

As far as I can see, it's the same bug.

Maybe time for a new full release? I've used the 2.0.46p4 build for quite a while now, no new issues that I can remember.